microsoft/EdgeML #17: Refactoring out serialization code

sureshiyengar, 2017-09-15:
Move the serialization code out of Bonsai model. One option is to have a serialization base class and different strategies for custom export which takes in a Bonsai model.

adityakusupati, 2017-09-16:
Hi @sureshiyenger,

Can you be more specific regarding the features you are talking about? Can you point me to the file and line number of the same?

Aditya

sureshiyengar, 2017-09-26:
Check void BonsaiModel::exportModel

Ideally, all the serialization logic can go into a separate class. The model class does not change if a new serialization format needs to be added.

adityakusupati, 2017-09-27:
Seems like a good idea. Will look into it.

sureshiyengar, 2017-09-27:
One thought I had was having a SerializationBase which defines the interface. The function takes in the BonsaiModel as output and serializes it into different format, more like a strategy pattern.

closed 2019-04-23 by adityakusupati