microsoft/EdgeML #29: Bonsai for CIFAR10

srplsnh, 2017-11-30:
Hi,

Couple of questions on applying Bonsai on CIFAR10, referring to your paper @ http://harsha-simhadri.org/EdgeML/files/Bonsai.pdf 

1. Table 1 refers to a dataset called CIFAR10-2. What does the -2 signify and is this dataset the same as 
the one from https://www.cs.toronto.edu/~kriz/cifar.html ?
The reason I ask is because Table 1 lists the number of features as 400 whereas CIFAR10 features should 32x32x3=3072.

2. Would it be possible for you to share Bonsai training parameters for CIFAR10, similar to the USPS scripts in your repository ?

Thanks !

adityakusupati, 2017-11-30:
Hi,

1) -2 implies that it is a binarised dataset of the CIFAR10. The datasets couldn't be released along with repo, sorry for the inconvenience. There was some pre-processing involved in getting feature dims to 400.
2) You would need the exact dataset we used for the params to reproduce the results, let me see what can be done.

srplsnh, 2017-11-30:
Thanks. On re-reading the paper I realized the title of the table does mention the naming convention you use. My bad, sorry. 
Just to confirm my understanding, by binarizing a dataset, do you mean deriving a 2-class dataset from an otherwise multi-class dataset ?

adityakusupati, 2017-11-30:
Yes, the # of data points remain the same, the classes are clubbed to make it a 2 class problem.

srplsnh, 2017-11-30:
I see. Did you try classifying the original 10-class CIFAR dataset with Bonsai ?

adityakusupati, 2017-12-01:
I myself, haven't tested it on the original CIFAR10 dataset. Let me know if you tried it and are facing any issues.

srplsnh, 2017-12-01:
In fact, I did make 2 preliminary runs just to test the waters, by changing 2 hyperparams (rest are the same as USPS):

1. Tree depth:6, Projected dim: 28, Final Test Accuracy: 0.438
2. Tree depth: 8, Projected dim: 128, Final Test Accuracy: 0.3695

The only preprocessing done was to normalize the input features to zero mean/unit variance.

Any suggestions on tuning hyperparameters and/or alternative preprocessing that could improve results would be appreciated.

adityakusupati, 2017-12-03:
Sorry for the delay. 
The code inherently does Zero Mean - unit variance normalisation, you don't need to feed normalised data and by exploring we found that it is the best normalisation for bonsai.
The hyper-params were specific to USPS dataset, so I can't expect them to run for CIFAR-10. Bonsai struggles for image datasets, which can be seen in CIFAR-2 and CHARS datasets. Also, Bonsai is for resource-constrained scenarios, so any depth greater than 3 will be a memory overhead unless you have very high sparsity going through. Also, Bonsai is constructed so as to work with lower dimensions, so when you have 128 dims then the projection matrix Z has the dimensions of 3072 x 128 which even with 90% sparsity accounts for 300KB.
It would be difficult for me to suggest the exact hyperparams, without looking at the patterns myself.

On the side not a possible pre-processing is to take features of neural-net(CNN) trained on CIFAR-10 and train Bonsai on those features, this will help you with prediction in Resource-constrained scenarios with a tolerable model-size and the features have induced separability.

closed 2017-12-26 by adityakusupati

ifed-ucsd, 2019-03-12:
Can you comment on which classes where grouped to form the binary datasets?

adityakusupati, 2019-03-15:
Hi,

I was trying to find the class grouping information, but it looks like the datasets the paper used were from an earlier paper called LDKL (http://manikvarma.org/pubs/jose13.pdf). I will try to get back to you either way. In general, you could always binarize the datasets anyway you wish and retrain the models. It would be great if you could write to me directly so that I can give you limited time access to all the datasets which can be downloaded.  Unfortunately, I can't release it publicly due to Microsoft policies, but please write to me (t-vekusu@microsoft.com) for all the datasets.