microsoft/EdgeML #31: tracking macosx support

jacobrosenthal, 2017-12-19:
Ive been sitting on the sidelines hoping someone would tackle osx support but it hasnt happened yet, so heres my start

```
Jacobs-Air:~ jacobrosenthal$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.3.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
Jacobs-Air:~ jacobrosenthal$ 
```

Yeah were clang masquerading on macosx, need gcc5

```brew install gcc5```
I already have it
```
Jacobs-Air:~ jacobrosenthal$ gcc-5 --version
gcc-5 (Homebrew GCC 5.5.0) 5.5.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

ah config.mk already uses with CC=g++-5

try to make

```
Jacobs-MacBook-Air:EdgeML jacobrosenthal$ make -j
In file included from utils.h:7:0,
                 from utils.cpp:4:
pre_processor.h:7:17: fatal error: mkl.h: No such file or directory
```

MKL doesnt appear to be in brew yet
was problematic on osx? maybe not now?
https://github.com/Homebrew/homebrew-science/issues/2840
I had to get that manually
https://software.intel.com/en-us/get-started-with-mkl-for-osx

thus in config.mk
MKL_ROOT=/opt/intel/mkl
and $(MKL_ROOT)/lib/intel64 becomes $(MKL_ROOT)/lib
and  /opt/intel/mkl/lib/intel64/ becomes /opt/intel/mkl/lib/ everywhere

bunch of this
```
/opt/intel/mkl/include/mkl_service.h:30:20: fatal error: stdlib.h: No such file or directory
```

You probably wont need to, but my brewed gcc5 was broken, and I had to 
```brew remove gcc5 && brew install gcc5```

Then in src/common/goldfoil.cpp feenableexcept isnt available 
Some stuff about fixing that but ill comment it for now
https://stackoverflow.com/questions/247053/enabling-floating-point-interrupts-on-mac-os-x-intel

Then I get this shitshow which I cant read at all
```
/var/folders/hv/syzcx3w10850txk2z6tbnb9m0000gn/T//ccsNs2oT.s:60644:11: note: change section name to "__const"
        .section __DATA,__const_coal,coalesced
                 ^      ~~~~~~~~~~~~
g++-5 -o ../../libProtoNN.so -shared -fPIC ProtoNNModel.o ProtoNNHyperParams.o ProtoNNParams.o ProtoNNTrainer.o ProtoNNPredictor.o ProtoNNFunctions.o cluster.o -lc 
Undefined symbols for architecture x86_64:
  "EdgeML::loadMinMax(Eigen::Matrix<float, -1, -1, 0, -1, -1>&, Eigen::Matrix<float, -1, -1, 0, -1, -1>&, int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)", referenced from:
      EdgeML::ProtoNN::ProtoNNPredictor::normalize()    in ProtoNNPredictor.o
  "EdgeML::saveMinMax(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)", referenced from:
      EdgeML::ProtoNN::ProtoNNTrainer::normalize()    in ProtoNNTrainer.o
  "EdgeML::l2Normalize(Eigen::SparseMatrix<float, 0, long long>&)", referenced from:
[... 179 lines cut ...]
      EdgeML::ProtoNN::ProtoNNPredictor::RBF()    in ProtoNNPredictor.o
      EdgeML::sparsekmeans::computeClosestCenters(Eigen::SparseMatrix<float, 0, long long> const&, float const*, unsigned long long, Eigen::Matrix<float, -1, -1, 0, -1, -1>&, unsigned long long*, float*) [clone ._cilk_for_fn.1] in cluster.o
      EdgeML::sparsekmeans::lloydsIter(Eigen::SparseMatrix<float, 0, long long> const&, unsigned long long, float const*, Eigen::Matrix<float, -1, -1, 0, -1, -1>&, unsigned long long*, bool) [clone ._cilk_for_fn.4] in cluster.o
      EdgeML::densekmeans::computeClosestCenters(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, float const*, long long, float*, unsigned long long*, float*) [clone ._cilk_for_fn.6] in cluster.o
      EdgeML::densekmeans::computePointsL2Sq(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, float*) [clone ._cilk_for_fn.8] in cluster.o
      EdgeML::densekmeans::lloydsIter(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, float const*, Eigen::Matrix<float, -1, -1, 0, -1, -1>&, unsigned long long*, bool) [clone ._cilk_for_fn.10] in cluster.o
      EdgeML::sparsekmeans::updateMinDistSqToCenters(Eigen::SparseMatrix<float, 0, long long> const&, float const*, unsigned long long, float*, float*, float*) in cluster.o
      ...
  "_cblas_sgemm", referenced from:
      EdgeML::sparsekmeans::distsqAllpointsToCenters(Eigen::SparseMatrix<float, 0, long long> const&, float const*, unsigned long long, float*, float const*, float*) in cluster.o
      EdgeML::densekmeans::distsqAllpointsToCenters(Eigen::Matrix<float, -1, -1, 0, -1, -1> const&, float const*, unsigned long long, float const*, float const*, float*) in cluster.o
      EdgeML::sparsekmeans::kmeanspp(Eigen::SparseMatrix<float, 0, long long> const&, float const*, Eigen::Matrix<float, -1, -1, 0, -1, -1>&) in cluster.o
  "_cblas_sgemv", referenced from:
      EdgeML::ProtoNN::ProtoNNPredictor::scoreDenseDataPoint(float*, float const*) in ProtoNNPredictor.o
      EdgeML::ProtoNN::ProtoNNPredictor::scoreSparseDataPoint(float*, float const*, unsigned long long const*, unsigned long long) in ProtoNNPredictor.o
  "_mkl_scsrmm", referenced from:
      EdgeML::sparsekmeans::distsqAllpointsToCenters(Eigen::SparseMatrix<float, 0, long long> const&, float const*, unsigned long long, float*, float const*, float*) in cluster.o
      EdgeML::sparsekmeans::kmeanspp(Eigen::SparseMatrix<float, 0, long long> const&, float const*, Eigen::Matrix<float, -1, -1, 0, -1, -1>&) in cluster.o
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
make[1]: *** [../../libProtoNN.so] Error 1
make: *** [libProtoNN.so] Error 2

```
Can Anyone read that mess?

harsha-simhadri, 2017-12-25:
@jacobrosenthal 
Sorry for the mess on mac. None of us have a macbook to replicate this; please bear with me while I find one to test.

jacobrosenthal, 2018-06-11:
Im able to use on mac via docker with:
```
# had to max out all resources to successfully build
FROM ubuntu:16.04

RUN apt-get update && apt-get -y install gcc g++ wget git make cpio && \
  wget -q -O - http://registrationcenter-download.intel.com/akdlm/irc_nas/tec/11544/l_mkl_2017.3.196.tgz | tar -xzf - && \
  cd l_mkl_2017.3.196 && \
  sed -i 's/ACCEPT_EULA=decline/ACCEPT_EULA=accept/g' silent.cfg && \
  ./install.sh -s silent.cfg && \
  cd / && \
  git clone https://github.com/Microsoft/EdgeML.git && \
  cd EdgeML && \
  make -Bj
ENV MKL_PATH=/opt/intel/mkl/lib/intel64_lin
ENV EDGEML_ROOT=/EdgeML
ENV LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$MKL_PATH:$EDGEML_ROOT
```
and then I make bash scripts for each command at /usr/local/bin/ that bring over working directory
```
#!/bin/bash
  
docker run -v $(pwd):/workspace -w /workspace edgeml:latest /EdgeML/BonsaiPredict "$@"
```

closed 2018-10-14 by harsha-simhadri