microsoft/EdgeML #49: Feature extractor script for HAR is buggy.

metastableB, 2018-08-20:
The the data produced at the end does not match the data used in publication in terms of the number of data points at least.

chogba, 2018-08-20:
[#47](https://github.com/Microsoft/EdgeML/pull/47) resolves this. The number of test data points is consistent. To get consistent number of points in val and train sets, set seed to 42 before shuffling subjects in `generateIndicesForSplits` in helpermethods.py

chogba, 2018-08-20:
[#47](https://github.com/Microsoft/EdgeML/pull/47) resolves this. The number of test data points is consistent. To get consistent number of points in val and train sets, set seed to 42 before shuffling subjects in `generateIndicesForSplits()` in helpermethods.py

closed 2018-08-20 by metastableB