microsoft/EdgeML #53: EMI-RNN notebooks error with "Cannot take the length of Shape with unknown rank."

jacobrosenthal, 2018-08-29:
```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-6-a98b2336be21> in <module>()
     12     print("y_cap shape is:", y_cap.shape)
     13 
---> 14     emiTrainer(y_cap, y_batch)

/Users/jacobrosenthal/Downloads/EdgeML/tf/edgeml/trainer/emirnnTrainer.py in __call__(self, predicted, target)
    101             assert self.trainOp is not None
    102             return self.lossOp, self.trainOp
--> 103         self.__validateInit(predicted, target)
    104         assert self.__validInit is True
    105         if self.graph is None:

/Users/jacobrosenthal/Downloads/EdgeML/tf/edgeml/trainer/emirnnTrainer.py in __validateInit(self, predicted, target)
     76     def __validateInit(self, predicted, target):
     77         msg = 'Predicted/Target tensors have incorrect dimension'
---> 78         assert len(predicted.shape) == 4, msg
     79         assert predicted.shape[3] == self.numOutput, msg
     80         assert predicted.shape[2] == self.numTimeSteps, msg

/opt/conda/lib/python3.6/site-packages/tensorflow/python/framework/tensor_shape.py in __len__(self)
    482     """Returns the rank of this shape, or raises ValueError if unspecified."""
    483     if self._dims is None:
--> 484       raise ValueError("Cannot take the length of Shape with unknown rank.")
    485     return len(self._dims)
    486 

ValueError: Cannot take the length of Shape with unknown rank.
```

metastableB, 2018-08-29:
Hi @jacobrosenthal,  could you please,

1. Provide your tensorflow and python versions
2. Provide a minimal working example

The notebooks we have pushed have been tested on python3 and Tensorflow 1.6+. I'm unsure where this bug is originating. `y_predicted` used in `emiTrainer.py:78` - the line causing the `ValueError`, should have a clearly defined rank of 4. The bug is being propagated from somewhere else.

jacobrosenthal, 2018-08-29:
1. Im using jupyter/tensorflow-notebook:latest docker image which would appear to be 8ccdfc1da8d5 which appears to have Python 3.6.3 and tensorflow (1.5.1)
2. Yeah not my code the notebook examples
3. Same error on all 3 EMI RNN notebooks. 

```
jovyan@483f90a6ff20:/Users/jacobrosenthal/Downloads/EdgeML/tf$ python --version
Python 3.6.3 :: Anaconda, Inc.
jovyan@180b3e5bcf13:/Users/jacobrosenthal/Downloads/EdgeML/tf$ pip list
DEPRECATION: The default format will switch to columns in the future. You can use --format=(legacy|columns) (or define a format=(legacy|columns) in your pip.conf under the [list] section) to disable this warning.
absl-py (0.4.0)
alembic (0.9.9)
appdirs (1.4.3)
asn1crypto (0.24.0)
async-generator (1.10)
attrs (18.1.0)
Automat (0.0.0)
backcall (0.1.0)
backports.weakref (1.0.post1)
beautifulsoup4 (4.6.3)
bleach (1.5.0)
bokeh (0.12.16)
certifi (2018.8.13)
cffi (1.11.5)
chardet (3.0.4)
cloudpickle (0.5.3)
conda (4.5.8)
constantly (15.1.0)
cryptography (2.2.1)
cycler (0.10.0)
Cython (0.28.5)
dask (0.18.2)
decorator (4.3.0)
dill (0.2.8.2)
entrypoints (0.2.3)
fastcache (1.0.2)
gmpy2 (2.0.8)
h5py (2.7.1)
html5lib (0.9999999)
hyperlink (17.3.1)
idna (2.7)
imageio (2.3.0)
incremental (17.5.0)
ipykernel (4.8.2)
ipython (6.5.0)
ipython-genutils (0.2.0)
ipywidgets (7.2.1)
jedi (0.12.1)
Jinja2 (2.10)
jsonschema (2.6.0)
jupyter-client (5.2.3)
jupyter-core (4.4.0)
jupyterhub (0.9.2)
jupyterlab (0.34.0)
jupyterlab-launcher (0.13.1)
Keras (2.1.6)
kiwisolver (1.0.1)
llvmlite (0.23.0)
Mako (1.0.7)
Markdown (2.6.11)
MarkupSafe (1.0)
matplotlib (2.2.3)
mistune (0.8.3)
nbconvert (5.3.1)
nbformat (4.4.0)
networkx (2.1)
notebook (5.6.0)
numba (0.38.1)
numexpr (2.6.6)
numpy (1.13.3)
olefile (0.45.1)
packaging (17.1)
pamela (0.3.0)
pandas (0.23.4)
pandocfilters (1.4.2)
parso (0.3.1)
patsy (0.5.0)
pexpect (4.6.0)
pickleshare (0.7.4)
Pillow (5.2.0)
pip (9.0.3)
prometheus-client (0.3.0)
prompt-toolkit (1.0.15)
protobuf (3.6.0)
ptyprocess (0.6.0)
pyasn1 (0.4.4)
pyasn1-modules (0.2.1)
pycosat (0.6.3)
pycparser (2.18)
pycurl (7.43.0.2)
Pygments (2.2.0)
pygpu (0.7.6)
pyOpenSSL (18.0.0)
pyparsing (2.2.0)
PySocks (1.6.8)
python-dateutil (2.7.3)
python-editor (1.0.3)
python-oauth2 (1.0.1)
pytz (2018.5)
PyWavelets (0.5.2)
PyYAML (3.12)
pyzmq (17.1.2)
requests (2.19.1)
ruamel-yaml (0.15.44)
scikit-image (0.14.0)
scikit-learn (0.19.2)
scipy (1.1.0)
seaborn (0.9.0)
Send2Trash (1.5.0)
service-identity (17.0.0)
setuptools (40.0.0)
simplegeneric (0.8.1)
six (1.11.0)
SQLAlchemy (1.2.11)
statsmodels (0.9.0)
sympy (1.1.1)
tensorflow (1.5.1)
tensorflow-tensorboard (1.5.1)
terminado (0.8.1)
testpath (0.3.1)
Theano (1.0.2)
toolz (0.9.0)
tornado (5.1)
traitlets (4.3.2)
Twisted (18.7.0)
urllib3 (1.23)
vincent (0.4.4)
wcwidth (0.1.7)
webencodings (0.5)
Werkzeug (0.14.1)
wheel (0.31.1)
widgetsnbextension (3.2.1)
xlrd (1.1.0)
zope.interface (4.5.0)
You are using pip version 9.0.3, however version 18.0 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
jovyan@180b3e5bcf13:/Users/jacobrosenthal/Downloads/EdgeML/tf$ 
```

adityakusupati, 2018-08-30:
@jacobrosenthal , we were able to recreate the bug. It is solely due to your tensorflow version being not compatible. TF made some major change during the transition from tf 1.5 to tf 1.6. Our code bases are currently supporting only >= tf 1.6.0. So please upgrade your tensorflow version and let us know if the problem persists. I was able to recreate the same error and was able to fix it by just changing the tf version.

Will be closing this now. Let us know if you still face issues.

closed 2018-08-30 by adityakusupati

jacobrosenthal, 2018-08-30:
Thank you for confirming.