microsoft/EdgeML #70: How to porting the the algorithm to run at Arduino or Microbit

skylin008, 2018-11-07:
Hello, Development Team! Thank you shared the algorithm ,but  how to porting the algorithm to run at arduino or microbit board! Thanks!

adityakusupati, 2018-11-10:
@skylin008 , thanks for your interest. The porting is very straight forward and all it takes is to write inference code in C and burn it onto device. In general we write hand written codes, but currently there is a compiler in works which will take in the model and spit out optimized devices prediction code with quantized weights. Simply put, just write a .ino file with your "C" code of inference be it ProtoNN, Bonsai, FastGRNN and link it to a model.h file and you are good to go.

You can follow this file as a reference: https://github.com/Microsoft/EdgeML/blob/master/Applications/GesturePod/onMKR1000/onMKR1000.ino

closed 2018-11-10 by adityakusupati

skylin008, 2018-11-13:
Thanks kindly!