microsoft/EdgeML #71: Could not open 'C:/Users/123/AppData/Local/Microsoft/Windows/UPPS/UPPS.bin' when open the EdgeML.sln

DengXueCheng, 2018-12-21:


adityakusupati, 2018-12-22:
I think it needs to be USPS, not UPPS. Please check and let us know.

DengXueCheng, 2018-12-24:
> I think it needs to be USPS, not UPPS. Please check and let us know.

Thank you for your answer. It's really a problem about a file called UPPS. bin. However, this problem has no effect on the normal operation of the program.Now I have trained the model. But ,how can I reduce the size of this model and deploy it into my own embedded system?

adityakusupati, 2018-12-24:
If you are using either Bonsai (https://github.com/Microsoft/EdgeML/blob/master/cpp/docs/README_BONSAI_OSS.md) or ProtoNN (https://github.com/Microsoft/EdgeML/blob/master/cpp/docs/README_PROTONN_OSS.ipynb) you can follow their readmes to play with the hyperparameters like sparsity, projection dimension to reduce the model size and then deploy it. Let me know any specific question you might have in this context.

DengXueCheng, 2018-12-25:
After training usps10, loadableModel：64kb, loadableMeanVar:3kb.
 I want to run them on ARM and make predictions. But at present, according to the code you provide, I can only run them on the computer.
I don't even know the models' encoding format. How can I put them into my embedded devices for prediction? And how to get to BonsaiOpt model?
If the model can be simplified into several matrices, I input X, and after a few simple matrix multiplication operations, the prediction results can be obtained. In this way, I can easily deploy it on embedded devices.
thank you

adityakusupati, 2018-12-25:
Go ahead and open the model files and see the formatting, there are both binary and readable model files. You can use readable model files to extract the model for your deployment as you said. Go into the model folder and you will find the appropriate matrices in Params directory. https://github.com/Microsoft/EdgeML/blob/master/cpp/docs/README_BONSAI_OSS.md has everything written in the Output section. 
Yes, we haven't released the Arduino code as we plan to automate the generation using a custom compiler, but you can always write the simple C code for prediction and then burn it onto ARM device as it follows the same algo as BonsaiPredictor.
Getting to BonsaiOpt model involves byte quantization and replacing tanh with its piecewise linear approximation. Nothing else is required to get to BonsaiOpt model. The issue with Arduino codes for varying models is the case of overflow and underflow, hence we haven't released handwritten codes for any of the datasets.

Let me know if your query regarding separate matrices and burning onto ARM and is resolved

adityakusupati, 2018-12-26:
If you do the math, you would see appending 1 at the end of the feature vector is the same as having a bias term for entire Bonsai classifier. It is just another way of adding bias and the 257th element is always 1 if you notice.