microsoft/EdgeML #75: Can you shed some light on your ICML paper experiments with CUReT and Chars datasets?

ifed-ucsd, 2019-02-28:
Looking at your paper "Resource-efficient Machine Learning in 2 KB RAM for the Internet of Things," it isn't clear to me how you set up the experiments for CUReT and Chars. 

For example, for CUReT, Table 1 lists the total number of images as 4204 + 1403 = 5607, but if one downloads the dataset from http://www.robots.ox.ac.uk/~vgg/research/texclass/, the number of images is 5612.

The setup for the Chars dataset is also unclear to me. Table 1 lists the total number of images as 4397 + 1886 = 6283, but downloading the dataset from http://www.ee.surrey.ac.uk/CVSSP/demos/chars74k/, the number of images is 7705.

Can you please specify how you ran these experiments such that we can have a fair comparison with the algorithm proposed in your paper?

adityakusupati, 2019-03-04:
@ifed-ucsd, it would be great if you could write to the first author, Ashish Kumar (https://ashishkumar1993.github.io/), of Bonsai paper directly as he created the datasets' splits along with features. There might be some filtering that could have been done during featurization.

adityakusupati, 2019-03-15:
@ifed-ucsd, as mentioned in the other issue thread, it would be great if you could write to me personally (t-vekusu@microsoft.com) and I could provide you with the exact datasets for download.

closed 2019-03-15 by adityakusupati