microsoft/EdgeML #93: SeeDot Example Error

FoxMarts, 2019-05-30:
Hello, Development Team! I was trying the new SeeDot tool, following the example provided, but when I use the command:
python SeeDot.py -a protonn --train ../../tf/examples/ProtoNN/usps10/train.npy --test ../../tf/examples/ProtoNN/usps10/test.npy --model ../../tf/examples/ProtoNN/usps10/output -o arduino
give me a error.
I am using environment python 3.5
Packages: Antrl4  4.7.2, Numpy  1.16.2, Scikit-learn 0.20.3, jupyter 1.0.0, numpy 1.14.5, pandas 0.23.4, scipy 1.1.0, tensorflow 1.12.2.

Ubuntu 16.04 with Gcc 1.7.4 and make 4.1.

The Output of the command:

================================
Executing on protonn for Arduino
--------------------------------
Train file: ../../tf/examples/ProtoNN/usps10/train.npy
Test file: ../../tf/examples/ProtoNN/usps10/test.npy
Model directory: ../../tf/examples/ProtoNN/usps10/output
================================

-----------------------
Collecting profile data
-----------------------
Generating input files for float training dataset...done

Build...success
Execution...success
Accuracy is 94.787%

-------------------------------------------------
Performing search to find the best scaling factor
-------------------------------------------------

Generating input files for fixed training dataset...done

Testing with max scale factor of 0
Generating code...failed!

Testing with max scale factor of -1
Generating code...completed
seedot_fixed.cpp: In function ‘int seedotFixed(MYINT**)’:
seedot_fixed.cpp:44:69: error: invalid conversion from ‘int’ to ‘const MYINT* {aka const short int*}’ [-fpermissive]
  SparseMatMul(&Widx[0], 256, X, 128, 128, &tmp5[0][0], 128, &Wval[0]);
                                                                     ^
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 2 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MYINT shrB, MYINT shrC);
      ^~~~~~~~~~~~
seedot_fixed.cpp:44:69: error: invalid conversion from ‘int’ to ‘MYINT* {aka short int*}’ [-fpermissive]
  SparseMatMul(&Widx[0], 256, X, 128, 128, &tmp5[0][0], 128, &Wval[0]);
                                                                     ^
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 4 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MYINT shrB, MYINT shrC);
      ^~~~~~~~~~~~
seedot_fixed.cpp:44:43: error: invalid conversion from ‘MYINT* {aka short int*}’ to ‘MYINT {aka short int}’ [-fpermissive]
  SparseMatMul(&Widx[0], 256, X, 128, 128, &tmp5[0][0], 128, &Wval[0]);
                                           ^~~~~~~~~~~
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 6 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MYINT shrB, MYINT shrC);
      ^~~~~~~~~~~~
seedot_fixed.cpp:44:61: error: invalid conversion from ‘const MYINT* {aka const short int*}’ to ‘MYINT {aka short int}’ [-fpermissive]
  SparseMatMul(&Widx[0], 256, X, 128, 128, &tmp5[0][0], 128, &Wval[0]);
                                                             ^~~~~~~~
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 8 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MYINT shrB, MYINT shrC);
      ^~~~~~~~~~~~
seedot_fixed.cpp:51:10: error: invalid conversion from ‘const MYINT* {aka const short int*}’ to ‘MYINT* {aka short int*}’ [-fpermissive]
   MatSub(&B[i][0][0], 1, &tmp6[0][0], 1, 25, 128, 1, &tmp5[0][0]);
          ^~~~~~~~~~~
[... 113 lines cut ...]
  File "SeeDot.py", line 94, in <module>
    obj.run()
  File "SeeDot.py", line 89, in run
    obj.run()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 325, in run
    return self.runForFixed()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 260, in runForFixed
    res = self.findBestScalingFactor()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 185, in findBestScalingFactor
    res = self.performSearch()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 136, in performSearch
    Common.Version.Fixed, Common.DatasetType.Training, Common.Target.X86, i)
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 118, in runOnce
    acc = self.predict(version, datasetType)
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/main.py", line 106, in predict
    acc = obj.run()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/predictor.py", line 128, in run
    acc = self.execute()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/predictor.py", line 109, in execute
    return self.executeForLinux()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/predictor.py", line 102, in executeForLinux
    acc = self.readStatsFile()
  File "/home/nelsonmartins/EdgeML/Tools/SeeDot/seedot/predictor.py", line 121, in readStatsFile
    return float(stats[0])
IndexError: list index out of range

sridhargopinath, 2019-05-31:
Hi,

Thanks for your interest in the tool.

I see that you are using Python 3.5. The tool uses a python dictionary feature that works on Python 3.6 and above. I can work on the fix to support python 3.5. In the meantime, can you test with Python 3.6?

Thanks,
Sridhar

FoxMarts, 2019-05-31:
Thanks for the quick response. I already experimented with python 3.6 and it worked.

closed 2019-05-31 by FoxMarts