microsoft/EdgeML #95: How to predict using bonsai or protoNN in python

FoxMarts, 2019-06-06:
Hello, I trained a model in Bonsai and a ProtoNN using the python codes presented in the examples. I would like to know if there is a way to only predict results in python with the trained model.

adityakusupati, 2019-06-10:
@FoxMarts, there are a couple of easy ways to do that. https://github.com/Microsoft/EdgeML/blob/master/tf/examples/Bonsai/bonsai_example.py#L80 -> this is the trained Bonsai Object and you could just write a simple prediction graph by calling it something like this: https://github.com/microsoft/EdgeML/blob/master/tf/edgeml/trainer/bonsaiTrainer.py#L58. This was written in order to facilitate only prediction. You can init a bonsai model through ckpt file or loading the numpy matrices using https://github.com/microsoft/EdgeML/blob/master/tf/edgeml/trainer/bonsaiTrainer.py#L274 and initializing Bonsai graph using https://github.com/Microsoft/EdgeML/blob/master/tf/edgeml/graph/bonsai.py#L9 as the constructor. 

Lastly, the Bonsai graph is very straight forward and the final prediction can be implemented in native python using numpy just by ensuring the tree prediction is preserved instead of the continuous tree formulation. Let me know if there is any issue.

FoxMarts, 2019-06-21:
Thanks for the quick and clear answer, I managed to work as a child. And sorry for the late reply.

closed 2019-06-21 by FoxMarts

nniranjhana, 2020-04-28:
Hi @FoxMarts @adityakusupati I tried to predict for a test set with this additional code in bonsai_example.py:

```
score, X_ = bonsaiObj(X = Xtest, sigmaI = 1e9)
correctPrediction = tf.equal(tf.argmax(tf.transpose(score), 1), tf.argmax(Ytest, 1))
acc = tf.reduce_mean(tf.cast(correctPrediction, tf.float32))
print (sess.run(acc))
```

But I keep getting the error:
```
tensorflow.python.framework.errors_impl.InvalidArgumentError: You must feed a value for placeholder tensor 'sigmaI' with dtype float
	 [[node sigmaI (defined at /ubc/ece/home/kp/grads/nniranjhana/TensorFledge/EdgeML/tf/edgeml_tf/trainer/bonsaiTrainer.py:56) ]]
```
How can I get the accuracy individually?

adityakusupati, 2020-04-28:
sigmaI is not a scalar, but rather a tensor. Please see this to see how to use it https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/trainer/bonsaiTrainer.py#L389. You can't use = for placeholders.

nniranjhana, 2020-04-29:
Oh thanks! @adityakusupati So how do I pass it from here? I tried declaring a tf.float for sigma but it didn't work either. How do I get accuracy alone?

adityakusupati, 2020-05-05:
This is no tf 2.xx, so you need to run session.run something like this https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/trainer/bonsaiTrainer.py#L462 - https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/trainer/bonsaiTrainer.py#L467

"score, X_ = bonsaiObj(X = Xtest, sigmaI = 1e9)" this will not work for Tf 1.xx as TF works on static graphs. EdgeML doesn't support to 2.x. But in case you like the functionality of tf 2.x you can use pytorch variants of edgeml which should work the way you wrote.