microsoft/EdgeML #99: SeeDot Question

FoxMarts, 2019-07-08:
Hi, I'm using protonn with the seedot tool for my C program but I have some doubts regarding the tool.
1. in my program I am using 32bits variables so I changed in common.py to WordLength from 16 to 32 and in the datatypes.h in the predictor folder I changed to #define INT32. I just wanted to know if anything else needed to be changed.
2. The size of the exponential tables is constant for 16bits and 32bits gives to change size in the code, in article vi it ​​was the variable T that has value 6 to read a value of 6 bits.
3. Continuing in the exponential I realized the theory in the article, but in the code, I do not understand the calculations of the variables tmp12, temp13, temp14.
Currently, the precision of the C model is very close to the python model even though it ignores 20 bits of the 32bits variables in the exponential calculation but wanted to try to figure out these calculations to try to implement a third table to cover more of the variable (greater precision?). and maintain the size of other exponential tables.

adityakusupati, 2019-07-08:
@sridhargopinath please take care of this issue.

sridhargopinath, 2019-07-16:
Hi @FoxMarts 

Thanks for your interest in SeeDot. Please find my responses below.

1. There is one more place you need to change: in the Arduino sketch. In `seedot/arduino/config.h`, switch the #define for MYINT.

2. That is correct. But the size of the table (the variable T) is configurable. To change the variable, go to `seedot/util.py` and edit `expBigLength`. I have tested the tool for values 3 through 6. 

3. Given two pre-computed tables EXP13A and EXP13B, and a negative input tmp11, the exponentiation algorithm works as follows.
    1. The input tmp11 is converted to a positive number and a bias term is subtracted.
    2. Twelve significant bits are extracted and stored in tmp12 and tmp13. The first six significant bits are stored in tmp12 and the next six bits are stored in tmp13.
    3. tmp12 and tmp13 are used to index into the two pre-computed tables EXP13A and EXP13B respectively.
    4. The final result is the product of the values from the two pre-computed tables.

    Since we are computing exponentiation of negative numbers, the result is still precise even though our exponentiation algorithm ignores a large number of bits. The idea here is that the output of exponentiation of negative numbers varies very little with a large change in input values. For positive numbers, the algorithm would need to be redesigned.

    Adding the third table would definitely improve precision. To do so, go to `seedot/compiler/ir/irBuilder.py `and edit the `populateExpTable` function. The formula to populate the third table is similar to the other two tables. And at runtime, you would need to extract six more bits from the input, index into the third table, and multiply the value with the values from the other tables.

Hope this helps.

Thanks,
Sridhar

FoxMarts, 2019-07-22:
Thank you for the answer, it helped me a lot.

closed 2019-07-22 by FoxMarts