microsoft/EdgeML #104: Fixing pre-processing script for google speech datasets

adityakusupati, 2019-07-15:
@metastableB , in the process_google.py file, I think the right way to create multiple versions of the datasets with different classes should be based on the classes listed in the labelmap rather than including everything and combining the not-required classes to 0. I think this should be chaged to support, say, simple things like google-30 where the classes are only the 30 keywords and no noise.

https://github.com/microsoft/EdgeML/blob/pytorch/pytorch/examples/SRNN/process_google.py#L35.

metastableB, 2019-07-17:
@adityakusupati you are taking care of this right?

adityakusupati, 2019-07-19:
@metastableB, I didn't get time yet. I will fix this today or otherwise, I think I will leave it as is.

closed 2019-07-19 by adityakusupati