microsoft/EdgeML #119: IHT routine failing due to numpy call on torch tensor

pushkalkatara, 2019-08-17:
IHT routine fails in BonsaiTrainer due to use of np.copy() on Torch tensor.

```
Epoch Number: 32

Classification Train Loss: 0.1960460032439894
Training accuracy (Classification): 0.9822221928172641
Test accuracy 0.931241
MarginLoss + RegLoss: 0.18808786571025848 + 0.13880659639835358 = 0.32689446210861206

Epoch Number: 33
Traceback (most recent call last):
  File "bonsai_example.py", line 94, in <module>
    main()
  File "bonsai_example.py", line 89, in main
    dataDir, currDir)
  File "/home/pushkalkatara/pytorch/lib/python3.5/site-packages/edgeml-0.2.1-py3.5.egg/edgeml_pytorch/trainer/bonsaiTrainer.py", line 337, in train
  File "/home/pushkalkatara/pytorch/lib/python3.5/site-packages/edgeml-0.2.1-py3.5.egg/edgeml_pytorch/trainer/bonsaiTrainer.py", line 129, in runHardThrsd
  File "/home/pushkalkatara/pytorch/lib/python3.5/site-packages/numpy/lib/function_base.py", line 792, in copy
    return array(a, order=order, copy=True)
  File "/home/pushkalkatara/pytorch/lib/python3.5/site-packages/torch/tensor.py", line 458, in __array__
    return self.numpy()
RuntimeError: Can't call numpy() on Variable that requires grad. Use var.detach().numpy() instead.
```

pushkalkatara, 2019-08-18:
https://github.com/microsoft/EdgeML/pull/120

adityakusupati, 2019-08-18:
@SachinG007 I though you tested all the example scripts. Can you please explain this?

pushkalkatara, 2019-08-22:
Solved and merged - #120 . Closing.

commit bd49385a (2019-08-21): IHT Routine Fix #119
closed 2019-08-22 by pushkalkatara