microsoft/EdgeML #121: Redundant fastTrainer?

adityakusupati, 2019-08-21:
[fastTrainer](https://github.com/microsoft/EdgeML/blob/harsha/reorg/edgeml_pytorch/trainer/fastTrainer.py). and [fastcell_example](https://github.com/microsoft/EdgeML/blob/harsha/reorg/examples/pytorch/FastCells/fastcell_example.py) of the harsha/reorg branch seem to be out of date and need to be updated or removed.

@harsha-simhadri I think thes file needs to be removed once your fastmodel is robust.

harsha-simhadri, 2019-08-21:
@adityakusupati The first link looks broken

adityakusupati, 2019-08-22:
It was because of the reorg - https://github.com/microsoft/EdgeML/blob/harsha/reorg/pytorch/edgeml_pytorch/trainer/fastTrainer.py

harsha-simhadri, 2019-08-22:
> It was because of the reorg - https://github.com/microsoft/EdgeML/blob/harsha/reorg/pytorch/edgeml_pytorch/trainer/fastTrainer.py

@adityakusupati Can you test the fastmodel in PR 123, and see if you are happy. If everything you need is there, I will remove the fastTrainer and fastcell_example files.

adityakusupati, 2019-08-22:
@harsha-simhadri  thanks for the response. I will do the needful and @SachinG007  will assist me. 
I think we need to change the names of the new files (both the trainer and example) for Fastcell just to be consistent.

adityakusupati, 2019-08-26:
- [ ] Arg parse should contain at least: dataDir, cellType, inputDims, hiddenDims, numEpochs, batchSize, learningRate, wRank, uRank, wSparsity, uSparsity, gate and update non-linearities, decayStep, decayRate and outputFile. Please follow one of [this](https://github.com/microsoft/EdgeML/blob/master/pytorch/examples/FastCells/helpermethods.py#L80)or [this](https://github.com/microsoft/EdgeML/blob/master/tf/examples/FastCells/helpermethods.py#L80) for default values of the argparse. 
- [ ] Mean-var normalization
- [ ] Create Model directory to save .npy models and .ckpt
- [ ] Save Mean and Std in model directory
- [ ] Save the command which has been executed in the model directory for future reference
- [ ] Provide support for all the Custom RNN cells in rnn.py to be used as part of our trainer and example 
- [ ] Save params in .npy format for the best model in the trainer which is updated as the epochs go by.
- [ ] Aggregate all the results in one txt file per cell type in the dataset directory. Use [this](https://github.com/microsoft/EdgeML/blob/master/pytorch/pytorch_edgeml/trainer/fastTrainer.py#L352) as reference. This dump contains, model size, model accuracy and the absolute path to the model directory and is very useful for grid searches and budget tuning.
- [ ] Complete dense training (for no sparsity)
- [ ] IHT for the 2nd phase with interleaved fixed support training (generally 1 IHT step for 15 fixed support train batches)
- [ ] 3rd phase with fixed support training
- [ ] Print the final best model numbers and the current model number along with the model directory at the end of the training.
- [ ] Please support my version of [quantSigm](https://github.com/microsoft/EdgeML/blob/master/pytorch/pytorch_edgeml/graph/rnn.py#L23) along with others to ensure stable quantization.
- [ ] Support batch_first=True and False for RNN unrolling.
- [ ] Support tuple input to RNN in case the cell is LSTM
- [ ] Make num_matrices a property so that people can access it outside for some other things.

I will add if I find any other thing, but this seems complete.