microsoft/EdgeML #129: SeeDot: X86 generated code (32-bit) gives segmentation fault

eirikvaa, 2019-09-13:
I'm working on using the X86 backend instead of the Arduino backend for the SeeDot compiler. I'm able to generate C++ code, but I get a segmentation fault on the following line in `seedot_fixed.cpp`:

```c++
SparseMatMul(&Zidx[0], &Zval[0], X, &tmp6[0][0], 257, 128, 128, 2);
```

If I instead use 256, such that the line then becomes

```c++
SparseMatMul(&Zidx[0], &Zval[0], X, &tmp6[0][0], 256, 128, 128, 2);
```

then it works.

If we then assume that I have generated an executable called `main` and run this with the following command

```c++
./main protonn fixed training
```

then I get the following output

```
#test points = 2007
Correct predictions = 352
Accuracy = 17.539
```

although, at the end of the training, I'm presented with the following output

```
Execution...success
Accuracy is 89.537%
```

Some questions:

- What is the reason for the segmentation fault?
- How come the accuracy is low (the first I mentioned), and not equal to the second I mentioned?
- I might have made some mistakes on the road to generating X86 and would be very glad for any directions to how to do his correctly. I had to make several changes to `SeeDot.py` and `main.py` in the `seedot` directory, as it was quite hardcoded for the Arduino platform.

sridhargopinath, 2019-09-17:
Hi @eirikvaa,

Thank you for your interest in SeeDot.

Could you please provide the exact command line used to generate code so that I can reproduce the error at my end? Also, did the errors occur after you modified `SeeDot.py` and `main.py`, or with the unmodified code?

Thanks,
Sridhar

eirikvaa, 2019-09-17:
> Could you please provide the exact command line used to generate code so that I can reproduce the error at my end?

For the ProtoNN implementation I have used exactly the same command line arguments as in the [tutorial](https://github.com/microsoft/EdgeML/tree/master/Tools/SeeDot) provided.

> Also, did the errors occur after you modified `SeeDot.py` and `main.py`, or with the unmodified code?

The error did indeed occur after I had modified `SeeDot.py` and `main.py`.

* [Diff](https://www.diffchecker.com/CMZ9fsle) of `main.py` vs `main86.py`
* [Diff](https://www.diffchecker.com/75pcUs9x) of `SeeDot.py` vs `SeeDot86.py`

I am sure that I have introduced some error along the way, but not sure where.

sridhargopinath, 2019-09-18:
It is hard for me to go through the diff to figure out the error. I have another suggestion. After SeeDot explores multiple programs, it will generate x86 code for the testing set in a temporary directory and executes that to obtain the accuracy of the fixed-point code. You can find this code in `TEMPDIR/seedot_fixed.cpp`. You can modify this code accordingly.

eirikvaa, 2019-09-18:
I can understand that. Ok, I will try that and report back, thank you very much for helping me out.

eirikvaa, 2019-09-18:
Everything works! I think it just boils down to copying the wrong set of files into the output directory. Thank you very much for your help.

closed 2019-09-18 by eirikvaa