microsoft/EdgeML #130: SeeDot: Example Arduino code always predict label '9'

eirikvaa, 2019-09-16:
When going through the Arduino/ProtoNN tutorial and running it on the device it seems to always predict label '9'.

```
------------------------
Average prediction time:
35994.77
------------------------

601: Predicted label: 9; True label: 9; Correct prediction
602: Predicted label: 9; True label: 9; Correct prediction
603: Predicted label: 9; True label: 9; Correct prediction
604: Predicted label: 9; True label: 9; Correct prediction
605: Predicted label: 9; True label: 9; Correct prediction
606: Predicted label: 9; True label: 9; Correct prediction
607: Predicted label: 9; True label: 9; Correct prediction
608: Predicted label: 9; True label: 9; Correct prediction
609: Predicted label: 9; True label: 9; Correct prediction
610: Predicted label: 9; True label: 9; Correct prediction
611: Predicted label: 9; True label: 9; Correct prediction
612: Predicted label: 9; True label: 9; Correct prediction
613: Predicted label: 9; True label: 9; Correct prediction
614: Predicted label: 9; True label: 9; Correct prediction
615: Predicted label: 9; True label: 9; Correct prediction
616: Predicted label: 9; True label: 9; Correct prediction
617: Predicted label: 9; True label: 9; Correct prediction
618: Predicted label: 9; True label: 9; Correct prediction
619: Predicted label: 9; True label: 9; Correct prediction
620: Predicted label: 9; True label: 9; Correct prediction

(...)
```

Is this a sign of a correctly functioning example? Am I missing something?

sridhargopinath, 2019-09-17:
The documentation on using the Arduino sketch is present here:
https://github.com/microsoft/EdgeML/blob/master/Tools/SeeDot/seedot/arduino/config.h

I will highlight some of the things relevant to this issue. By default, the Arduino prediction code runs only on a single data point since the entire data set cannot fit on the device. This behavior is used to compute the total prediction time of the generated code. There is one more mode called the accuracy mode, where the entire dataset is streamed to the device using serial communication. In this setting, the generated code sequentially runs on each entry in the data set. This mode is useful to compute the accuracy of the generated code.

Since you are running the prediction time mode (default mode), the generated code is being run on only a single data point.

Hope this description helps.

Regards,
Sridhar

eirikvaa, 2019-09-17:
Oh, of course, that makes total sense. Thank you very much :-) I'm closing the issue because of your good response.

eirikvaa, 2019-09-18:
I have modified `config.h` to run in accuracy mode by doing the following.

```
#define ACCURACY
//#define PREDICTIONTIME
```

Is there something else I must do? Nothing is happening in the serial because of `syncWithMaster();` in the `accuracy` function.

If I send `fixed` as a message to the Serial port then it writes out `point`, but nothing is happening.

sridhargopinath, 2019-09-18:
In the accuracy mode, you need to send each data point to the device using serial communication. There is a master, typically a desktop machine, that has access to the data set and is responsible for sending each data point to the client, the Arduino device. The first step in the Arduino code is to sync with the master. After that, each data point is sent as a stream of bytes. The `getIntFeature` in `arduino.ino` reads the stream of bytes and converts them to integers.

You would need to write code for the master to make this work.

eirikvaa, 2019-09-18:
Ok, that was the answer I was looking for – I will try to make it work.

closed 2019-09-18 by eirikvaa