microsoft/EdgeML #145: ModuleNotFoundError: No module named 'edgeml.trainer'

ccu1tn, 2019-11-19:
Dear every one; 

When I run this command python bonsai_example.py -dir usps10/. This source appears a this error 
[image]
I'm looking way how to fix it like uses pip install module named 'edgeml.trainer'.  But It is not done. 
so. What is the module I need to install for this errors?

Many thanks!

adityakusupati, 2019-11-19:
Hi,

I think you are trying to run the tensorflow version bonsai_example.py after installing pytorch version of the edgeml module. It would be great if you can attach the sequence of instructions you have followed for the same. 

Aditya

ccu1tn, 2019-11-19:
Dear @adityakusupati 
Thank you so much.. It worked.

closed 2019-11-19 by harsha-simhadri