microsoft/EdgeML #155: How to create new input from device sensors for prediction [Arduino Sketch]

robroe-tsi, 2020-02-23:
Hi,

**I used SeeDot to generate code (for a Bonsai model) for the Arduino device and now I want to change the sketch so that new data (read from sensors of the device) is prepared as input features and fed into the prediction method. My question is how to do this?**

It looks like the feature values for the predict() method need to be available in an array X. The sample feature vector that is part of the generated model.h file (PROGMEM MYINT X) contains only integers. But the raw sensor values that where used in the model training script in python (from train.npy) where floats. So there seems to be some ETL (standard., quanti.) that transforms the raw values into those suitable for the model in c. It looks like this ETL is spread over different scripts - some stuff is done in helpermethods.py other later with SeeDot
Do I need to rewrite this ETL in C completely? Or is is already part of the generate C files?

Remark: this question is similar to https://github.com/microsoft/EdgeML/issues/109 but the latter was closed without any answer.

I appreciate any help. Many thanks and best, Roberto

ShishirPatil, 2020-05-23:
@sridhargopinath Please look into this.

sridhargopinath, 2020-05-29:
@aayan636

affot, 2022-01-19:
Hello @robroe-tsi 

did you find any answer on your question?
Currently I am also struggling with the usage of the models that have been created.

Regards
affot