microsoft/EdgeML #158: AttributeError: 'FastGRNNCell' object has no attribute 'num_weight_matrices'

theoxo, 2020-02-28:
Hi,

While running the PyTorch example for FastCells, I got the following error:

```
Traceback (most recent call last):
   File "fastcell_example.py", line 94, in <module>
     main()
   File "fastcell_example.py", line 82, in main
     device=device)
   File "path/EdgeML/pytorch/edgeml_pytorch/trainer/fastTrainer.py", line 50, in __init__
     self.numMatrices = self.FastObj.num_weight_matrices
   File "path/miniconda3/envs/pytorch/lib/python3.6/site-packages/torch/nn/modules/module.py", line 576, in __getattr__
     type(self).__name__, name))
AttributeError: 'FastGRNNCell' object has no attribute 'num_weight_matrices'
```

Making the `num_weight_matrices` field visible externally, e.g. by adding `self.num_weight_matrices = self._num_weight_matrices` in the \_\_init\_\_ of FastGRNNCell appears to resolve the issue.

adityakusupati, 2020-02-28:
@theoxo  thank you for pointing this out. We have fixed this in https://github.com/microsoft/EdgeML/pull/159

Please check and let me know.

theoxo, 2020-02-29:
Oops, I should have checked the PRs first! #159 works, thank you.

adityakusupati, 2020-02-29:
Thanks. Let us close this issue then.

closed 2020-02-29 by theoxo