microsoft/EdgeML #168: How to get the accuracy of the bonsai_example.py in TF?

nniranjhana, 2020-04-26:
After training the model, in the bonsai_example.py in TF, I want to separately print the accuracy of the model. How do I do that? Should I invoke the `accuracyGraph()` function before `sess.close()` in bonsai_example.py?

nniranjhana, 2020-04-28:
Could anyone help with this? In ProtoNN example, it is simple to call `ProtoNN.accuracy` to get it, but it doesn't seem to be straightforward with the Bonsai example.

adityakusupati, 2020-05-04:
https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/trainer/fastTrainer.py#L480

There is an accuracy graph built for Bonsai where you can feed in the data and get the accuracy.  Please refer to the example in the above-mentioned line.

closed 2020-12-16 by SachinG007