microsoft/EdgeML #175: Seedot generated arduino sketch(Bonsai) for usps-10 dataset not producing any output when uploaded on arduino M0

mehreenjabeen, 2020-05-04:
Hi,
I changed some parameters while training Bonsai on usps-10 dataset and obtained model of size < 2KB. The resulting model i gave input to seedot.py and obtained arduino sketch. This sketch when run on Arduino M0 is uploading perfectly but I cant see any output.  What could be the possible reason.
Baudrate I have set to 115200.

Sketch uses 17712 bytes (6%) of program storage space. Maximum is 262144 bytes.
Global variables use 3300 bytes of dynamic memory

adityakusupati, 2020-05-04:
@sridhargopinath please look at this.

sridhargopinath, 2020-05-04:
HI @mehreenjabeen Mehreen,

Can you please give me more details to reproduce the error?

1. What are the hyperparameters used to train Bonsai on usps-10?
2. What command line did you use to invoke SeeDot?

With these details, I can try to reproduce the error on my end.

mehreenjabeen, 2020-05-04:
Hi,
I used the below command:
python3 bonsai_example.py -dir usps10/ -d 2 -p 5 -rW 0.001 -rZ 0.0001 -rV 0.001 -rT 0.001 -sZ 0.1 -sW 0.1 -sV 0.2 -sT 0.2 -e 100 -s 1

mehreenjabeen, 2020-05-04:
Hi,
I used the below commands:
   python3 bonsai_example.py -dir usps10/ -d 2 -p 5 -rW 0.001 -rZ 0.0001 -rV 0.001 -rT 0.001 -sZ 0.1 -sW 0.1 -sV 0.2 -sT 0.2 -e 100 -s 1

python3 SeeDot.py -a bonsai --train <path/to/train.npy> --test <path/to/test.npy> --model <path/to/Bonsai/model>

sridhargopinath, 2020-05-05:
Hi Mehreen,

I tried your configuration on Arduino Uno instead of the M0 and it seems to work as expected for me. I was able to produce the following output on the serial monitor:
```
1: Predicted label: 4; True label: 9; WARNING: Incorrect prediction
2: Predicted label: 4; True label: 9; WARNING: Incorrect prediction
3: Predicted label: 4; True label: 9; WARNING: Incorrect prediction
...
```

I guess it might be an issue with the device you are working on. But just to be sure, can you share the standard output of SeeDot? If the execution is successful, you will see the following output:
```
------------------------------
Generating code for arduino...
------------------------------

Generating input files for fixed testing dataset...done

Generating code...completed

Arduino sketch dumped in the folder /tmp/SeeDot/2020-05-05_09-37-05/arduino

```
If SeeDot is successful, then there might be minor issues with the device-specific code present at `/tmp/` folder. Since I haven't worked with the M0, can you check if it uses different pragmas than the Uno? You can also check if M0 can produce output on simple sketches. Let me know.

Hope this helps.

mehreenjabeen, 2020-05-12:
Hi Sridhar,

Apologies for replying late. 
I got the same output as yours for SeeDot. The issue seems with the M0(couldn't run basic sketches on it too). I converted the generated arduino sketch to cpp code and it ran perfectly on desktop. 

Thanks,

closed 2020-05-12 by mehreenjabeen