microsoft/EdgeML #178: SeeDot: Arduino Accuracy or Prediction Mode & Documentation

BrianEubanks, 2020-05-19:
Is there any documentation on using the SeeDot generated Arduino sketch? The [README](https://github.com/microsoft/EdgeML/blob/master/tools/SeeDot/seedot/arduino/README.md) is empty.

I'm having issues specifically using Accuracy mode so I can test a whole data set and not be limited to the one preset data point in Prediction mode.

I trained a Bonsai model using USPS10 data and then created the Arduino sketch using SeeDot. I was following all the instructions up to this point and was getting good results, however I cannot get any predictions to work with accuracy mode.

> In the accuracy mode, you need to send each data point to the device using serial communication.
There is a master, typically a desktop machine, that has access to the data set and is responsible for sending each data point to the client, the Arduino device. The first step in the Arduino code is to sync with the master. After that, each data point is sent as a stream of bytes. The `getIntFeature` in `arduino.ino` reads the stream of bytes and converts them to integers.

> You would need to write code for the master to make this work.

_Originally posted by @sridhargopinath in https://github.com/microsoft/EdgeML/issues/130#issuecomment-532715765_

Could you explain this a little more for me? After I send "fixed" to sync the master, I don't think I'm sending the data in the right format. My understanding is that the data points are stored in the CSV files that SeeDot created, and I should be able to send it a data point from the X.csv file and the prediction should match whats in the Y.csv file. However I've been unable to get any kind of result.

adityakusupati, 2020-05-21:
@sridhargopinath can you look at this.

sridhargopinath, 2020-05-29:
@aayan636