microsoft/EdgeML #179: accuracyGraph output

AbdelilahTAHIRI, 2020-05-21:
Hello, 

I am using Bonsai model from tf package.

I am looking for model tuning using Bayesian optimization. To do so I need the accuracy of the model and its size. 

How can I get the accuracy from the model after one training. I tried the accuracyGraph() function but it returns a tensor object. Here is the code and the result:

```
depth=2
projection_dim=3
sparz=1.000000
sparw=0.100000
sparv=0.100000
spart=1.000000
sess = tf.InteractiveSession()
bonsaiObj = Bonsai(numClasses, dataDimension, projection_dim, depth, sigma)
bonsaiTrainer = BonsaiTrainer(bonsaiObj, regW, regT, regV, regZ, sparw, spart, sparv, sparz,
                          learningRate, X, Y, useMCHLoss, outFile)
sess.run(tf.global_variables_initializer())
bonsaiTrainer.train(batchSize, totalEpochs, sess,Xtrain, Xtest, Ytrain, Ytest, dataDir, currDir)
accuracy=bonsaiTrainer.accuracyGraph()
sess.close()
print(accuracy)

Non-Zero : 53.0 Model Size: 0.26171875 KB hasSparse: True

For Classification, Maximum Test accuracy at compressed model size(including early stopping): 0.99886984 at Epoch: 87
Final Test Accuracy: 0.9934074
Tensor("Mean_244:0", shape=(), dtype=float32)
```

I was expecting to have the Final Test Accuracy as an output.

Could anyone help with this please?

adityakusupati, 2020-05-22:
Hi,

This is not built for Tf 2.x, so it is a static graph. calling accuracy = self.accuracyGraph() will just set up the graph.

You need to run the graph to evaluate it with the appropriate data https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/trainer/bonsaiTrainer.py#L400. ALso, the accuracy graph is built already in the bonsaiTrainer. So all you need to do is sess.run([bonsaiTrainer.accuracy], _feed_dict=). Please check the syntax in the above link.

AbdelilahTAHIRI, 2020-05-26:
Looks working. 

Many thanks

closed 2020-05-26 by AbdelilahTAHIRI