microsoft/EdgeML #180: Training New Gestures on the GesturePod

mousetrapbot01, 2020-05-21:
Hello, I am trying to train new gestures for the GesturePod on the particle photon, a arduino-like mcu. I have some questions regarding training.

Do you guys have the raw data used to train the model data.h on the github repository?

What is the point of setting the toggle in the training pipeline to 0? Can the parts where the toggle is 0 not be discarded?

Also, the positive/negative examples used to train the model confuses me. If my understanding is correct, the positive examples were trained in a way where every gesture is performed within a 2 second slide, and then the user stops for a bit, in which case the label is 0, silence. Then the user performs it again. Why was the number of samples collected for twirl much higher 353 compared to the other gestures? Additionally, what did you mean by increasing the train set size by ten-fold by adding additional examples where the region is shifted by 25ms? 

Also, for the negative examples, from my understanding, users just walked around with the cane in hand and 8 minutes of that data was collected. However, I don't understand where its says you guys clipped it to generate 2-second segments of negative examples. How do you label those negative examples? Should the negative examples be considered as noise for the 2 second segment and silence for the rest? Another confusion was about the partial gestures. Should those examples also be labeled as noise?

For different data sizes, did the performance of the model not good enough until the last set of false positives were added?

Thank you.

ShishirPatil, 2020-05-21:
> Do you guys have the raw data used to train the model data.h on the github repository?

The GDPR compliant data set has been released, and can be accessed here - https://www.microsoft.com/en-us/research/uploads/prod/2018/05/dataTR_v1.tar.gz

> What is the point of setting the toggle in the training pipeline to 0? Can the parts where the toggle is 0 not be discarded?

Think of toggle as a coarse label that can then be used to finely label the data. The coarse label is collected during data-collection, and the fine labeling is done post-data-collection using the scripts for labeling. For example, the toggle can be _High_ for 5 seconds, and the gesture is a 2 second subset of the 5 seconds.

> positive/negative examples used to train the model

The number of gestures of twirl being higher is an artifact of the process. There is nothing significant about this. Note that 353 includes chirality (right twirl and left twirl).

> increasing the train set size by ten-fold

This is the idea of perturbation. Toy example: Consider a gesture that lasts 4 time stamps : [1,2,3,4,5,6,7,8,9,10] Here, {4,5,6,7} represent the gesture with {1,2,3,8,9,10} being the silence. Now I can generate multiple timestamp windows centered around the gesture {4,5,6,7}- a) {2,3,4,5,6,7,8}, b) {3,4,5,6,7,8,9}. The gestures only need to be approximately centered -there is some space to wriggle around. 

> users just walked around with the cane in hand and 8 minutes of that data was collected

Just use simple sliding windows to generate 2 second clips from the 8 minutes of data. Since, all the data is negative (no-gesture), just label all the 2 second windows as noise. 

>  partial gestures. Should those examples also be labeled as noise?

Yes. Labeling partial gestures as noise helps increase precision. 

> For different data sizes, did the performance of the model not good enough until the last set of false positives were added?

The performance of model was "good" (high recall) even before the last set of false positives were added. The last set of false positives helped make the model more "robust" (lower false-positives) in our real-world deployment. 

Hope this helps! Let me know if you have any further questions.

mousetrapbot01, 2020-05-22:
Thank you for your swift reply.

In the data that in the link, there is no column for the toggle field. Does that mean that the toggle column was erased in the process?

Also, are the files that you provided supposed to be added directly to the generateFeatures.py file to generate the train.csv and test.csv?

In the edgml protoNN training algorithm, I went to examples/tf/protonn and used the gerenated train.csv, test.csv to convert them into npy using genfromtxt('foo.csv', delimiter=','), then put the two files into the protoNN_example.py. Then I got matrices B,W,Z and gamma in a numpy format. Using these files, I modified the genDataHeader.py file so that it takes numpy files instead of tsv files by np.load, and ran the three files through, getting a new data.h. Is this the expected pipeline for training a new data.h?

Thank you.

ShishirPatil, 2020-05-23:
> In the data that in the link, there is no column for the toggle field. Does that mean that the toggle column was erased in the process?

Yes. Since we anyway have provided the precise labels, there is no value from the (less-precise) toggle.

> Also, are the files that you provided supposed to be added directly to the generateFeatures.py file to generate the train.csv and test.csv?

Yes!

> In the edgml protoNN training algorithm, I went to examples/tf/protonn and used the gerenated train.csv, test.csv to convert them into npy using genfromtxt('foo.csv', delimiter=','), then put the [...]

Yes, sounds about right.

> I am using the mpu9250, and so in creating the data-collection.ino

I don't see why this should be a problem. Let me know how it goes. I can take a look at it, if you run into issues.

mousetrapbot01, 2020-05-29:
Thank you for the quick reply.

I have some problems regarding the data that you provided. There, the columns do not match what the python script 'generateFeatures.py' suggests - 'mlabel' instead of the 'label' column. Were there updates to the code that require some changes to the data that is posted? Also the script seemed to need a format without the _labelled at the end. Is there a way to access the version that contains the script that will accept this data without needing modifications?

I am using particle photon and the mpu9250 for my project.

Also for the mpu9250, I was looking up libraries and found these two below,
https://github.com/jdvr1994/MPU9250PhotonLibrary - this one did not give me any errors, but it does not have any dmp features which is why I initially implemented my version of data-collection without dmp.  So far, I have tried collecting data for two features and adding some noise that was not processed. I noticed that when I trained the model, the accuracy was around 3%, which I suspect is some error to the format of the data.

https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library - this one does have the dmp functionality that is synonymous to the features that were implemented in gesturepod's version of serialcollection.ino. I was able to find adequate substitutes. However, the problem is the library breaks when I try to compile, which presents another problem in itself. (I suspect there is some bug from with particle's makefile/compiler) My current plan is to proceed using this library, and I will keep you updated.

closed 2020-10-17 by harsha-simhadri