microsoft/EdgeML #181: Load a saved model

AbdelilahTAHIRI, 2020-05-22:
Hi,

So as to evaluate a saved Bonsai model on the test set. I am using loadModel(currDir) which returns a dicts with all parameters and hyper-parameters.

My question is how can we feed all those learnt parameters (W, Z, V, and T) to a BonsaiTrainer?
 
Thanks for help

adityakusupati, 2020-05-22:
Once you the dictionary, you can use it to init the tensors in https://github.com/microsoft/EdgeML/blob/master/tf/edgeml_tf/graph/bonsai.py. 

loadModel is part of bonsaiTrainer, so all you need to do is use those params and load them into bonsaiObj which can be done in multiple ways. 

Or you can store a .ckpt file using tf;s internal model saver and load it back.

AbdelilahTAHIRI, 2020-05-26:
Many thanks for the feedback.

I didn't pay attention for the default attributes of the bonsaiObj class (Z, W, V and T).

Here is the snippet of my code. It worked well.

```
#Open an interactive session
sess = tf.InteractiveSession()

#instantiate a BonsaiTrainer
bonsaiObj = Bonsai(numClasses, dataDimension, projection_dim, depth, sigma)
bonsaiTrainer = BonsaiTrainer(bonsaiObj, regW, regT, regV, regZ,
                              sparw,spart, sparv, sparz,
                              learningRate, X, Y, useMCHLoss, outFile)
# save np.load
np_load_old = np.load
# modify the default parameters of np.load
np.load = lambda *a,**k: np_load_old(*a, allow_pickle=True, **k)

# Load Bonsai Model
paramDict, hyperParamDict=bonsaiTrainer.loadModel("./TFBonsaiResults/Bonsai_Tuning_Output/")

# restore np.load for future normal usage
np.load = np_load_old

# Model Learnt Params
W=paramDict['W']
V=paramDict['V']
T=paramDict['T']
Z=paramDict['Z'] 

# Model Hyper-parameters
dataDim=hyperParamDict['dataDim']
projDim=hyperParamDict['projDim']
numClasses=hyperParamDict['numClasses']
depth=hyperParamDict['depth']
sigma=hyperParamDict['sigma']

# Feed BonsaiTrainer instance with learnt Params
bonsaiObj = Bonsai(numClasses, dataDim, projDim, depth, sigma,W=W,Z=Z, V=V,T=T)
bonsaiTrainer = BonsaiTrainer(bonsaiObj, regW, regT, regV, regZ, sparw, spart, sparv, sparz,
                          learningRate, X, Y, useMCHLoss, outFile)
sess.run(tf.global_variables_initializer())

```

closed 2020-05-26 by AbdelilahTAHIRI

adityakusupati, 2020-05-27:
Great!