microsoft/EdgeML #182: SeeDot Compiler Errors:

AbdelilahTAHIRI, 2020-05-27:
Hi!

When running the SeeDot.py script to generate an arduino sketch, I got a lot of such errors:
```

seedot_fixed.cpp:46:67: error: invalid conversion from ‘int’ to ‘MYINT* {aka short int*}’ [-fpermissive]
  SparseMatMul(&tmp6[0][0], &Zidx[0], X, 128, &Zval[0], 16, 128, 10);
                                                                   ^
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 4 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MY
      ^
seedot_fixed.cpp:46:67: error: invalid conversion from ‘const MYINT* {aka const short int*}’ to ‘MYINT {aka short int}’ [-fpermissive]
  SparseMatMul(&tmp6[0][0], &Zidx[0], X, 128, &Zval[0], 16, 128, 10);
                                                                   ^
In file included from seedot_fixed.cpp:7:0:
library.h:18:6: note:   initializing argument 5 of ‘void SparseMatMul(const MYINT*, const MYINT*, MYINT**, MYINT*, MYINT, MYINT, MYINT, MYINT)’
 void SparseMatMul(const MYINT *Aidx, const MYINT *Aval, MYINT **B, MYINT *C, MYINT K, MYINT shrA, MY
      ^
seedot_fixed.cpp:50:62: error: invalid conversion from ‘const MYINT* {aka const short int*}’ to ‘MYINT* {aka short int*}’ [-fpermissive]
  MatSub(&mean[0][0], 64, 1, 1, 3, &tmp6[0][0], &tmp7[0][0], 1);
                                                              ^
In file included from seedot_fixed.cpp:7:0:
library.h:8:6: note:   initializing argument 1 of ‘void MatSub(MYINT*, const MYINT*, MYINT*, MYINT, MYINT, MYINT, MYINT, MYINT)’
 void MatSub(MYINT *A, const MYINT *B, MYINT *C, MYINT I, MYINT J, MYINT shrA, MYINT shrB, MYINT shrC
      ^
seedot_fixed.cpp:50:62: error: invalid conversion from ‘int’ to ‘const MYINT* {aka const short int*}’ [-fpermissive]
  MatSub(&mean[0][0], 64, 1, 1, 3, &tmp6[0][0], &tmp7[0][0], 1);

```
It seems that the generated code  perform calls tor library inline functions (`SparseMatMul(), MAtSub()` and so on) with wrong positioned arguments.

Can anyone help with this please

adityakusupati, 2020-05-29:
@sridhargopinath can you please take a look?

sridhargopinath, 2020-05-29:
@aayan636

aayan636, 2020-06-01:
@AbdelilahTAHIRI Please share the input code you used which generated this error.

AbdelilahTAHIRI, 2020-06-02:
I am training the bonsai on a custom dataset but not the ups10 set.  
@aayan636 here is the command line invoked 
`
python3.5 SeeDot.py -a bonsai --train  ../../examples/tf/Bonsai/DriveStyleData/Data/one_based_svm/train.txt --test ../../examples/tf/Bonsai/DriveStyleData/Data/one_based_svm/test.txt  --model ../../examples/tf/Bonsai/DriveStyleData/Data/one_based_svm/BonsaiResults/10_59_50_26_5/Params/   -o ../../examples/tf/Bonsai/DriveStyleData/SeeDotResults/
`
Despite these errors, the SeeDot compiler generates the code with mispositioned arguments when calling library functions from `predict `function. Below a snippet of the generated predict.cpp file:

```

int predict() {
	MYINT tmp15[3];
	MYINT tmp29;
	MYINT tmp23[3];
	MYINT node2;
	MYINT tmp22[1][1];
	MYINT tmp14[1][1];
	MYINT tmp6[3][1];
	MYINT tmp24[3][1];
	MYINT tmp8[3];
	MYINT tmp11[3][1];
	MYINT tmp9[3][1];
	MYINT tmp13[3];
	MYINT tmp7[3][1];
	MYINT tmp28[3][1];
	MYINT tmp19[3][1];
	MYINT tmp21[3];
	MYINT node0;
	MYINT tmp12[3][1];
	MYINT tmp20[3][1];
	MYINT tmp25[3];
	MYINT tmp16[3][1];
	MYINT tmp17[3];
	MYINT tmp26[3][1];
	MYINT tmp10[3];
	MYINT tmp27[3][1];
	MYINT node1;
	MYINT tmp18[3][1];

	// Z |*| X
	memset(tmp6, 0, sizeof(MYINT) * 3);
	SparseMatMul(10, &Zval[0], &tmp6[0][0], &Zidx[0], 128, 128, 16);

	// tmp6 - mean
	MatSub(&mean[0][0], 1, 1, 64, &tmp7[0][0], 1, &tmp6[0][0], 3);

	node0 = 0;

	// W * ZX
	MatMulCN(3, &tmp8[0], &tmp7[0][0], &tmp9[0][0], &W[node0][0][0], 128, 128, 3, 2, 1, 0);

	// V * ZX
	MatMulCN(128, 1, 2, 3, &tmp7[0][0], 0, &tmp10[0], 3, &tmp11[0][0], 128, &V[node0][0][0]);

	// tanh(V0)
	TanH(&tmp11[0][0], 1, 2, 3);

	// W0 <*> V0_tanh
	MulCir(&tmp11[0][0], 1, 2, 1, &tmp9[0][0], 3, &tmp12[0][0]);

	// T * ZX
	MatMulCN(&tmp7[0][0], 2, 3, &T[node0][0][0], 0, &tmp13[0], 1, 128, 128, 1, &tmp14[0][0]);

	node1 = ((tmp14[0][0] > 0) ? ((2 * node0) + 1) : ((2 * node0) + 2));

	// W * ZX
	MatMulCN(&tmp15[0], 3, 3, 128, 2, 0, 128, 1, &tmp16[0][0], &tmp7[0][0], &W[node1][0][0]);

	// V * ZX
	MatMulCN(128, 2, 3, 128, 1, &tmp17[0], 3, 0, &tmp18[0][0], &V[node1][0][0], &tmp7[0][0]);

	// tanh(V1)
	TanH(2, 1, 3, &tmp18[0][0]);

	// W1 <*> V1_tanh
	MulCir(3, 1, 2, &tmp18[0][0], &tmp19[0][0], 1, &tmp16[0][0]);

	// score0 + tmp19
	MatAdd(1, &tmp20[0][0], &tmp12[0][0], 1, &tmp19[0][0], 1, 1, 3);

	// T * ZX
	MatMulCN(3, &tmp21[0], 128, 0, 2, &tmp7[0][0], &T[node1][0][0], 128, 1, &tmp22[0][0], 1);

	node2 = ((tmp22[0][0] > 0) ? ((2 * node1) + 1) : ((2 * node1) + 2));

	// W * ZX
	MatMulCN(2, &tmp7[0][0], 128, 3, &tmp23[0], 128, 0, 3, &tmp24[0][0], 1, &W[node2][0][0]);

	// V * ZX
	MatMulCN(&tmp7[0][0], &tmp25[0], 3, &tmp26[0][0], &V[node2][0][0], 2, 3, 1, 128, 0, 128);

	// tanh(V2)
	TanH(1, &tmp26[0][0], 2, 3);

	// W2 <*> V2_tanh
	MulCir(&tmp27[0][0], 2, &tmp26[0][0], 1, &tmp24[0][0], 3, 1);

	// score1 + tmp27
	MatAdd(&tmp20[0][0], 1, &tmp27[0][0], 3, &tmp28[0][0], 1, 1, 1);

	// argmax(score2)
	ArgMax(&tmp28[0][0], &tmp29, 1, 3);

	return tmp29;
}

```

closed 2021-02-23 by SachinG007