microsoft/EdgeML #184: End to end quantization process for FastGRNN

marjanemd, 2020-06-12:
Hello,

I am applying the **FastGRNN** model on **Google-12 Dataset** and I am getting the same accuracy mentioned in the paper for floating values **with Sparsity and low rank matrices**. I have also quentized the weights and biases using the **"quantizeFastModels.py"**. I am trying to apply the quantized weights on my model, and I have some questions:

1. In the paper, in Byte Quantization part (section 3.3), it is mentioned that simple integer quantization of weights leads to a large loss in accuracy, so in the FastGRNN the weights (W,U) have been trained with regard to piecewise linear approximation of non-linear activation functions.

**Looking at the quantization process in "quantizeFastModels.py",** the quantization is applied with a **simple scaling factor,** which does not consider piecewise linear approximation of activations. I was wondering if the quantized weights giving  92% accuracy (based on the paper) are made from a different .py code.

2. I could not find **FastGRNN model evaluation** for **end to end quantized values**. Meaning there wa**s no scale or offset for the quantized input** or any right/left shift after metrice's multiplication.( input*weights will usually have some overflow values in quantized models )

Any help would be much appreciated.

adityakusupati, 2020-06-22:
@marjanemd I just got to this as I didn't get any notification about this issue. Sorry for the delay. 

1) Yes, the quantization is with a simple scaling factor and it works that way because we expect the users to train with the piecewise linear approximation of activations like quantTanh and quantSigm already supported in our code. The models are obtained from this same .py file and need to be handled well during inference to ensure all integer.operations. 

2) You are correct, we have to ensure the code written doesn't suffer from overflow in these cases. The inference code for quantized models is not same as the typical pytorch code. We need to write C code to ensure that all ops are integer ops. Please keep in mind that the working memory can be more than 1 byte so intermediate values can be larger than 1 byte, but will be scaled back for next operations.

I have emailed you the inference codes we used for reference for now.

marjanemd, 2020-06-22:
Thanks @adityakusupati for your response, and email.

For part 1, I tried to train my network with quantTanh and quantSigm, but it did not learn anything. Are there any specific steps to take before using these 2 activations?

For part 2, does that mean there is no way to create an end to end quantized model in python?

adityakusupati, 2020-06-22:
1) That is weird. It should work as well as the default activations. 
2) You can write similar inference code in python, it will be a couple of 10s of lines, similar to pytorch forward pass.

adityakusupati, 2020-06-26:
@marjanemd  can we close this issue if your doubts have been clarified?

marjanemd, 2020-06-26:
yes, Thanks for your great help!

closed 2020-06-30 by marjanemd