microsoft/EdgeML #210: [ASK] : Bonsai model on Azure IoT Edge

vermaaditya, 2020-10-12:
Hi Team, 

We are trying to deploy the Bonsai Models on a Machine in the plant hosting Azure IOT Edge runtime. 
Is there a reference documentation which can be leveraged? Or do you have a suggested deployment pattern to achieve the same on Azure IoT Edge in the form of Modules?

Kindly suggest.

Thanks,

SachinG007, 2020-10-12:
Hi Aditya,
Can you please check if you are referring to the correct Bonsai, this repository is for the ICML'17 publication Bonsai (https://github.com/microsoft/EdgeML/blob/master/docs/publications/Bonsai.pdf) , which is a strong non linear tree based classifier.

SachinG007, 2020-10-12:
Hi @vermaaditya ,
Can you please check if you are referring to the correct Bonsai, this repository is for the ICML'17 publication Bonsai (https://github.com/microsoft/EdgeML/blob/master/docs/publications/Bonsai.pdf) , which is a strong non linear tree based classifier.

vermaaditya, 2020-10-13:
Looks like an incorrectly posted question, thanks for clearing the confusion. The question was meant to be for - https://github.com/BonsaiAI/bonsaiai.github.io

Closing the issue.

closed 2020-10-13 by vermaaditya