microsoft/EdgeML #211: Time series regression problem

premkumar1303, 2020-10-15:
I have seen that data is flattened across timestep dimensions and label values are added in the first column of **train.npy** data and the **numClasses** is 10 for usps10 data which actually looks like a classification problem. I am trying to use the Fastcell algorithms for the time-series regression problem i.e. to forecast future values. In my case, I won't have the parameter like **numClasses**. So what should I assign for **numClasses** variable? Can these algorithms work for time-series regression problems? Waiting for the reply.

SachinG007, 2020-10-15:
Hi @premkumar1303 ,
Yes you can use FastCells for time series regression problems. numClasses parameter would be redundant in that case (in the usps example provided, numClasses is used for adding a FC layer over the last hidden state of FastCell).
A very easy way to build a regression with the code provided would be to modify the loss function [here](https://github.com/microsoft/EdgeML/blob/92f9654c0e0db76e751c44334e9d5428b445f34a/pytorch/edgeml_pytorch/trainer/fastTrainer.py#L93) from crossEntropy to a MSE.
Thanks

premkumar1303, 2020-12-03:
Hello @SachinG007 ,
I have done as you told. It worked fine for my regression problem. The classifier function in FastTrainer class is also redundant in this case. After training, we will get the parameters in .npy format. How can I save the model in .h5 file format so that we can load it and get the inference by giving the input. Please help me out.
Thank you.

SachinG007, 2020-12-03:
Hi @premkumar1303 ,
1) Can you please elaborate upon why do you need .h5 file format to save the model. If you want to run on a microcontroller, you can create a C header file with weights in static consts 
2) This link for .h6 conversion might be helpful : https://discuss.pytorch.org/t/save-torch-tensors-as-hdf5/39556

premkumar1303, 2020-12-03:
1. I want to compare the model size with tflite model which I have done for the same application.
2. Can you also please help me how to load these parameters as a model and get the inference for an input.

Thanks @SachinG007 .

premkumar1303, 2020-12-03:
Please tell me, Is it correct to compare the model size calculated in **FastTrainer.py** with the tflite model file?

SachinG007, 2020-12-16:
Hi @premkumar1303,
Sorry for the late response.
It might not be correct to compare the model size calculated in FastTrainer.py with tflite model file. FastTrainer.py file outputs the model size based on the parameters as floating-point (32 bits), whereas tflite does optimizations to the tf model like converting them to half-precision (16 bits) or 8-bit integers. Read more about tflite optimizations here: https://www.tensorflow.org/lite/guide/get_started
 
If you are fine with having floating point model, then you might not want to do quantization. However, if you want to anyways get a model size corresponding to the quantized fastGRNN, please follow the quantization instructions at the end of this Readme: https://github.com/microsoft/EdgeML/tree/master/examples/pytorch/FastCells and divide the model size calculated by FastTrainer.py by 4.

premkumar1303, 2021-01-03:
Hi @SachinG007,
Thank you for the information and sorry for the late response. The information provided is helpful.
During the model size calculation in FastTrainer.py, sW < 0.5 is taken. If I consider the sparsity of 0.5 and above for some k rank, we will get the same model size. Why only sW < 0.5 is considered, does it has any significance?

SachinG007, 2021-01-03:
Hi @premkumar1303,
It is because sparsity>0.5 doesn't lead to model compression. You need to store the positions of the nonzero values too. Hence better have no sparsity instead of s=0.6

premkumar1303, 2021-01-04:
Ok thank you.

premkumar1303, 2021-01-10:
Hello @SachinG007, 
Is there any way to deploy machine learning models on a raspberry pi other than using tflite. If yes can you please provide some references. 
And, how can I get inference by giving inputs using the weights saved from the trained FastRNN model?

Thank you.

closed 2021-02-23 by SachinG007