microsoft/EdgeML #216: Questions About RNN Pool

snakers4, 2020-12-14:
Hi,

My name is Alexander, we are mostly [working](https://github.com/snakers4/silero-models) on high-quality fast Speech Recognition. We mostly use strided 1D Convolutions and Transformer modules. We mostly use progressive strides (i.e. 2 - 2 - 2) in our models. I have tried overall stride of 2 and 4 and 8 was better (on a limited compute budget ofc). But I have not looked into other configurations, like 2 - 4 for example (very similar to an example you show in your blog post).

As you have noted in your blog [post](https://www.microsoft.com/en-us/research/blog/seeing-on-tiny-battery-powered-microcontrollers-with-rnnpool/), having an expressive stride module is of utmost importance. Your post made me think about improving our models further and I took the liberty of looking through your [code](https://github.com/microsoft/EdgeML/blob/master/pytorch/edgeml_pytorch/graph/rnnpool.py) briefly.

Though the idea in itself is kind of simple (just use a RNN to a batched input, in case of 1D it is even easier) a series of questions arose after taking a look at your code because it kind of heavily departs from PyTorch standard practices. So  I would like to know whether all of this is a bug or feature, so to say:

(0)
The module contains `.to(torch.device("cuda")` but it does not inherit from `FastGRNNCUDA` (it inherits from `FastGRNN`).
It seems a bit strange, so why is it the case?

(1)
You use `.to(torch.device("cuda")`, though it is a standard practice in PyTorch to write device agnostic code. Does this imply that:

- This code is NOT meant for multi-node (or multi-device) parallelization (e.g. DP , DDP)?
- This code is NOT meant to be run later on x86 (quantized or pruned) inference afterwards?

(2)
I saw some pruning and low-rank snippets in your code.
Low-rank does not seem to be used in RNNPool.

(3)
I took a quick look at `RNNCell` and `FastGRNNCell`. Apart from having some utilities for model size estimation, seeming unused low rank options and pruning, apart from param initialization I cannot really why you went for implementing these classes from scratch instead of just using the standard PyTorch ones. Is there some reasoning behind it?

(4)
```
The network is composed of a base VGG network followed by the
    added multibox conv layers.  Each multibox layer branches into
```
You seem to apply RNNPool to a VGG [encoder](https://github.com/microsoft/EdgeML/pull/215/files?file-filters%5B%5D=.py#diff-7f590a9d5d099bc186b312bff4a294bf7e7daef5aaa8c9078142e92b6d78d80cR51). VGG is usually slow and large, is there any reason you do not apply this scheme to a mobilenet?

snakers4, 2020-12-14:
Some of these are addressed by a unmerged [PR](https://github.com/microsoft/EdgeML/pull/215/files?file-filters%5B%5D=.py#diff-7f590a9d5d099bc186b312bff4a294bf7e7daef5aaa8c9078142e92b6d78d80cR91-R102), but nevertheless would be nice to hear from you

ShikharJ, 2020-12-15:
Hey @snakers4, thanks for your interest and mentioning your queries.

I can answer regarding (2) that the low-rank implementation was motivated from potential use in our exploratory analysis with this operator, which ultimately didn't make it into any of the experiments mentioned in the paper. Hence it's not currently used in any model scripts. But feel free to experiment with this implementation, should you foresee a suitable use case.

@oindrilasaha and @MJ10  Would be better suited to help with the rest of your queries.

oindrilasaha, 2020-12-16:
Hi Alexander, I can answer the rest of your concerns
(0) Ideally we would like to incorporate FastGRNNCUDA into RNNPool because it ensures faster training/inference. However we are facing few issues with this and are currently working on them.
(1) We used .to(torch.device("cuda") in tensor initialization in RNNPool since we are using cuda for training and inference for the sole purpose of speed only. You are welcome to change it to .to(torch.device("cpu") (or however you want to use it) if you want cpu only inference.
(3) FastGRNN is not available as a standard class in Pytorch and we need these classes to define the graph of FastGRNN and also to support functionalities like low-rank and sparsification.
(4) If you check the model files you can notice that we are indeed using MBConv blocks as was used in MobileNet. We state this in our paper also.

Thanks

snakers4, 2020-12-16:
@oindrilasaha @ShikharJ 

I see, many thanks for these answers!
We have just read your papers and they also clarified these things a bit.
We will  definitely be trying the proposed 2-4 downscale schemes albeit adapted to our domain.

What I like is this ablation test (i.e. you can be lazy and just use plain vanilla modules):

[image]

If you are interested I can post some ablations tests (i.e. baseline models with 2-2-2 vs a naive strided model with 2-4 vs RNNPool for 1D data) here once we are done.

SachinG007, 2020-12-16:
@snakers4 
Sure, it would be great to hear about your results in speech recognition using RNNPool

ShikharJ, 2021-06-04:
Closing because of inactivity.

closed 2021-06-04 by ShikharJ