microsoft/EdgeML #219: i want to use Bosani with anaconda spyder

musaosman78, 2020-12-17:
hi,
i have a problem when i am trying to use the Bosani module in my spyder 

  File "<ipython-input-2-34ffe291a199>", line 1, in <module>
    import helpermethods

ModuleNotFoundError: No module named 'helpermethods'
also if i want this model in my own dataset how can i use 
thanks in advance

SachinG007, 2020-12-18:
Hi @musaosman78,
1) helpermethods.py is in the same location as bonsai_example.py here: https://github.com/microsoft/EdgeML/tree/master/examples/pytorch/Bonsai
Can you please give more setup details so that I can help you? 

2) `bonsai_example.py` assumes that data is in a specific format.  It is
assumed that train and test data is contained in two files, `train.npy` and
`test.npy`. Each containing a 2D numpy array of dimension `[numberOfExamples,
numberOfFeatures + 1]`. The first column of each matrix is assumed to contain
label information.  For an N-Class problem, we assume the labels are integers
from 0 through N-1.

So either you can convert your dataset to the above format or else modify the preProcessData function [here](https://github.com/microsoft/EdgeML/blob/cbba9f8b989e545788427c004eb8450e7e4c1a21/examples/pytorch/Bonsai/helpermethods.py#L162) accordingly.

musaosman78, 2020-12-19:
...: np.random.seed(42)

#Loading and Pre-processing dataset for Bonsai
dataDir = "usps10/"
(dataDimension, numClasses, Xtrain, Ytrain, Xtest, Ytest, mean, std) = helpermethods.preProcessData(dataDir, isRegression=False)
print("Feature Dimension: ", dataDimension)
print("Num classes: ", numClasses)
Feature Dimension:  19
Num classes:  8
my label is binary classification why i get Num classes:  8 
how can i plot the accuracy and logloss function

musaosman78, 2020-12-19:
how to solve this error 
my dataset is binary dataset 
  lab_ = np.zeros((Xtrain.shape[0], numClasses))
MemoryError

musaosman78, 2020-12-19:
Thanks SachinG007 , another question how to plot the accuracy and the loss function

SachinG007, 2020-12-21:
Hi @musaosman78,
The line [here](https://github.com/microsoft/EdgeML/blob/5f0b6e86791018645f37e44c9c638b0e642b0ab9/pytorch/edgeml_pytorch/trainer/bonsaiTrainer.py#L356) gives you the training loss and training accuracy after every epoch. You can dump these values in a list and plot them later. You can also use [tensorboardX ](https://github.com/lanpa/tensorboardX) for monitoring the same values while the training is ongoing.

closed 2020-12-23 by SachinG007