microsoft/EdgeML #223: Bonsai model, classification report, and confusion matrix

musaosman78, 2020-12-22:
Hi, 
how can i get the classification report and the confusion matrix for bonsai model I wish any one help me with this issue

SachinG007, 2020-12-22:
Hi @musaosman78,
We will be more than happy to help you with any issue.
Classification report and confusion matrix for the bonsai model can be calculated in exactly the same way as it is done for any other classification model. The accuracy function [here](https://github.com/microsoft/EdgeML/blob/5f0b6e86791018645f37e44c9c638b0e642b0ab9/pytorch/edgeml_pytorch/trainer/bonsaiTrainer.py#L91) gives you the class predictions, using which you can easily build the confusion matrix or a classification report as explained [here](https://stackoverflow.com/questions/53290306/confusion-matrix-and-test-accuracy-for-pytorch-transfer-learning-tutorial).

Also, can I close issue #219

musaosman78, 2020-12-22:
sorry for my questions, but is it same for tensorflow (tf)

SachinG007, 2020-12-22:
Hi @musaosman78,
I am not sure if I understood your doubt correctly, but the classification report and confusion matrix is independent of whether you are using tf or pytorch for training. You just need the class predictions and the corresponding labels, irrespective of the library used.

closed 2020-12-29 by SachinG007