microsoft/EdgeML #226: Unable to download dataset

pankaja0285, 2021-01-05:
- I cloned the repo 
- In anaconda prompt installed all the required libraries in requirements.txt
- opened in pycharm on my desktop 
- tried to run download_mscoco.sh with edit configurations - set to folder for e.g. "coco-dataset" and click run, it just opens up the cmd prompt very quickly and closes up. The download doesn't happen. 

What am I doing wrong? Please advice.
[image]

SachinG007, 2021-01-06:
@oindrilasaha

oindrilasaha, 2021-01-08:
Hi, can you please ensure that you have set the script as executable?

closed 2021-02-23 by SachinG007