microsoft/EdgeML #227: MAdds mismatch problem

colycolycolyyo, 2021-01-12:
I am very interesting in this great work:RNNPool Effificient Non-linear Pooling for RAM Constrained Inference.
However,I met some problems when I reproduce the MAdds of resnet18 on the following chart：
[image]
It‘s 0.95G when the layer1.layer2 and maxpooling are replaced by rnnpooling layer.I reproduce the MAdds of RNNPooling layer of input size : (1,3,224,224) which is 0.334G(334430208) in the following approach:
[image]
Because most calculation about rnnpooling come from FastGRNNCell.The above picture is forward function of FastGRNNCell.I use the variable count  to accumulate the operation times.
Hence,I got the result MAdds of rnnpooling layer:0.334G(334430208).The counterpart MAdds of  resnet18(exclude layer 1,layer 2,maxpooling) is 0.91G(910880768). So could you please tell me how did you calculate the overall MAdds 0.95G of resnet18 with RNNPool layer?Thank you!

ShikharJ, 2021-01-12:
@oindrilasaha

oindrilasaha, 2021-01-12:
Hi, In ResNet18-RNNPool specified in the table you attached, we are replacing layers after the first convolution - namely the first maxpool layer and the 4 residual blocks after that. So the input to RNNPool is 112x112x64 and output is 28x28x128, therefore FLOPs being ~100MFLOPs. The ResNet layers replaced constitute ~900MFLOPs.

WUSHUANGPPP, 2021-01-12:
Thanks for your reply,oindrilasaha and ShikharJ!
I  replaced the layers you mentioned,maxpooling,4 residual blocks.The input size keep same.But I got the FLOPS results 334MFLOPs of RNNPooling layer all the same.I just sum up all the operations in FastGRNNCell forward function,just like the snapshot above. Did I do something wrong?@oindrilasaha

oindrilasaha, 2021-01-13:
Hi, the input to the RNNPoolLayer is 112x112x64. Also, WX calculation of RNN1 we do as input_height x input_width x input_channels x hidden_dimension reusing computations in future patches. Can you also tell me what are the hidden dimensions of RNN1 and RNN2 that you assume in your above approach?

WUSHUANGPPP, 2021-01-13:
> Hi, the input to the RNNPoolLayer is 112x112x64. Also, WX calculation of RNN1 we do as input_height x input_width x input_channels x hidden_dimension reusing computations in future patches. Can you also tell me what are the hidden dimensions of RNN1 and RNN2 that you assume in your above approach?
[image]
For sure,hidden dimensions of RNN1 & RNN2 are same,they are 32 in my practice.So the output dimensions being 4*32 = 128.My FLOPs calculation approach is different with yours,I guess this is the reason why MAdds mismatch...But I don't know specific reason. All the calculation operations happen in FastGRNNCell. So in my practice,the global variable _count_ to record all the matrix multiplication operations in FastGRNNCell.Then I print the variable count after the rnnpool layer,then I got 334M...

WUSHUANGPPP, 2021-01-15:
@oindrilasaha
Hi,Saha.I‘m still confuse about how you get 100MFLOPs.The lowest-level code is FastGRNNCell,all the Multiply-Adds happen in FastGRNNCell.The calculation formula —— input_height x input_width x input_channels x hidden_dimension which not consider the recurrent loops and stride repetitive calculation.Could you please tell me how you calculate 100MFLOPs specifically?

oindrilasaha, 2021-01-19:
Hi, I tried your counting code and I see the difference. Like I said, currently as the code is written, the WX computations of the first RNN are being repeated i.e. for column and row pass we need the same WX values but currently the code is calculating them twice. Also, since the patches overlap(by 4), the overlapping 4x8 pixels need not be calculated again but they are being recalculated in the current code. I am calculating RNN1  WX for reporting as 112x112x64x32. So, wherever you are doing count+= , if you print all the values and ignore all the 11943936 values printed and add up the rest of them with 112x112x64x32(~25MFLOPs) you will get 157MFLOPs(not 100,forgive the error).

WUSHUANGPPP, 2021-01-25:
Okey , thank you for your explanation！

closed 2021-01-25 by WUSHUANGPPP

ahdxwg, 2023-07-23:
Is there a function for computing flops? Don't know how to make the calculation