microsoft/EdgeML #229: Build Failed!!

premkumar1303, 2021-02-16:
Hi,
I have trained ProtoNN model on usps10 as given in the steps. Then when I am trying to run SeeDot.py, I am getting the message Build.. FAILED!!. I have gone through the predicor.py where at the line process = subprocess.call(args, stdout = file) is returning 1. What could be the reason and please help me with the solution.

Thank you.

ShikharJ, 2021-02-16:
@premkumar1303 Can you post the full output please?

premkumar1303, 2021-02-16:
[image]

ShikharJ, 2021-02-16:
@premkumar1303 Can you also post your gcc version please?

premkumar1303, 2021-02-16:
[image]

ShikharJ, 2021-02-16:
@premkumar1303 Are you directly building from terminal? My hunch is that this is a path issue on your machine.

premkumar1303, 2021-02-16:
@ShikharJ  Yes, I am directly building from the terminal. Is there any other way? I have added the GCC and MSBuild paths to environment variables. My doubt is that does this has any dependency on Microsoft visual studio?

ShikharJ, 2021-02-16:
@aayan636 Would be able to get a better hang of this. We've tried running the same command on our machine, and we're unable to reproduce this issue. So most likely it is a machine specific issue.

krantikiran68, 2021-02-16:
@premkumar1303 Could you please try running 
'''
mkdir -p temp
 python SeeDot.py -a protonn --train ../../examples/tf/ProtoNN/usps10/train.npy --test ../../examples/tf/ProtoNN/usps10/test.npy --model ../../examples/tf/ProtoNN/usps10/output -o arduino --tempdir ./temp/
cd temp/
'''
This will create the intermediate files in the temp directory and take you there. 
Now you can manually try building the files and see what the error is hopefully.

premkumar1303, 2021-02-16:
[image]

These are the list of files I am getting in the temp folder. 

```
C:\Users\ChintU\EdgeML\tools\SeeDot\temp>g++ -o a.exe seedot_fixed.cpp
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x71): undefined reference to `SparseMatMul(short const*, short const*, short**, short*, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0xb7): undefined reference to `MatSub(short*, short const*, short*, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x13e): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x1bc): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x1dc): undefined reference to `TanH(short*, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x21a): undefined reference to `MulCir(short*, short*, short*, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x298): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x343): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x3c1): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x3e1): undefined reference to `TanH(short*, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x41f): undefined reference to `MulCir(short*, short*, short*, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x465): undefined reference to `MatAdd(short*, short*, short*, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x4e3): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x58e): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x60c): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x62c): undefined reference to `TanH(short*, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x66a): undefined reference to `MulCir(short*, short*, short*, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x6b0): undefined reference to `MatAdd(short*, short*, short*, short, short, short, short, short)'
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: C:\Users\ChintU\AppData\Local\Temp\ccDQs69W.o:seedot_fixed.cpp:(.text+0x72b): undefined reference to `MatMulCN(short const*, short*, short*, short*, short, short, short, short, short, short, short)'
[... cut ...]
c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/bin/ld.exe: c:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.5.0/../../../../x86_64-w64-mingw32/lib/../lib/libmingw32.a(lib64_libmingw32_a-crt0_c.o): in function `main':
R:\winlibs64-7.5.0\mingw-w64-v7.0.0\mingw-w64-crt/crt/crt0_c.c:18: undefined reference to `WinMain'
collect2.exe: error: ld returned 1 exit status
```

Above one is the error which I am getting when building seedot_fixed.exe. I am a beginner at building files. Please correct me if I am wrong and provide any link which describes building files manually.

[image]

This is the file created in the subfolder.

krantikiran68, 2021-02-16:
The compilation has to be done for the files in the folder together. You can refer to the `Makefile` in the folder for reference. 
The `Predictor.vcxproj` would allow for the building of the entire folder like the Makefile. For that you may need to create a Visual Studio project as shown [here](https://docs.microsoft.com/en-us/cpp/build/working-with-project-properties?view=msvc-160).

krantikiran68, 2021-02-16:
A simpler way to try out SeeDot is by installing [WSL](https://docs.microsoft.com/en-us/windows/wsl/install-win10) with [VS Code](https://code.visualstudio.com/download) and using WSL to run SeeDot.

premkumar1303, 2021-02-16:
I will try it out and get back to you if I still face any issues.  Thank you for the information @krantikiran68, @ShikharJ.

ShikharJ, 2021-02-23:
@premkumar1303 I'm closing this issue for now. We can reopen this later if you are still facing issues.

closed 2021-02-23 by ShikharJ

premkumar1303, 2021-03-08:
Hello,
1. SeeDot is written for bonsai and protonn algorithms. Can I use the files for FastGRNN/FastRNN algorithms by doing some modifications? If yes, can you please give me an idea of what all the things to change?
2. In the [FastGRNN](https://github.com/microsoft/EdgeML/blob/master/docs/publications/FastGRNN.pdf) paper, prediction times on raspberry pi for different algorithms and datasets were given. How can I deploy and run FastGRNN algorithm on raspberry pi to make the predictions? I have trained the FastGRNN model for a dataset (private) on laptop to make predictions. I got the weigths. Now how can I deploy the model onto raspberry pi to make predictions?

Can you please give me an idea of how to achieve the above steps. Thanks in advance.