microsoft/EdgeML #230: USE_DROPOUT = True

ablanda, 2021-03-08:
Hi everyone, 
I'm trying to set USE_DROPOUT = True, but I get this error:
InvalidArgumentError: You must feed a value for placeholder tensor 'EMI/FastGRNN/keep-prob' with dtype float
	 [[{{node EMI/FastGRNN/keep-prob}}]]

I have run just your TensorFlow example, file 01_emi_fastgrnn_example.ipynb.

Could you help me, please?

ShikharJ, 2021-03-08:
@metastableb

metastableB, 2021-03-11:
Hi @ablanda, please post the stack trace of your error so that I can help you better. With the information I have, the error you seem to be having is from this [line](https://github.com/microsoft/EdgeML/blob/c7349dd67ede4e49d58460c46d3957b7879c299e/tf/edgeml_tf/graph/rnn.py#L1990)

```
  if self.useDropout is True:
                keep_prob = tf.placeholder(dtype=tf.float32, name='keep-prob')
                wrapped_cell = tf.contrib.rnn.DropoutWrapper(cell,
                                                             input_keep_prob=keep_prob,
                                                             output_keep_prob=keep_prob)
                outputs__, states = tf.nn.static_rnn(
                wrapped_cell, x, dtype=tf.float32)
                outputs = []
```

So `EMIFastGRNN` does initialize the `keep_prob` with a placeholder as desired by tensorflow as long as `self.useDropout` is true.

ShikharJ, 2021-04-13:
Closing for inactivity.

closed 2021-04-13 by ShikharJ