microsoft/EdgeML #231: FastGRNN

premkumar1303, 2021-03-11:
Hello,

1. SeeDot is written for bonsai and protonn algorithms. Can I use the files for FastGRNN/FastRNN algorithms by doing some modifications? If yes, can you please give me an idea of what all the things to change?
2. In the FastGRNN paper, prediction times on raspberry pi for different algorithms and datasets were given. How can I deploy and run FastGRNN algorithm on raspberry pi to make the predictions? I have trained the FastGRNN model for a dataset (private) on laptop to make predictions. I got the weigths. Now how can I deploy the model onto raspberry pi to make predictions?

Can you please give me an idea of how to achieve the above steps. Thanks in advance.

ShikharJ, 2021-03-11:
@aayan636 Can you please take a look at this?

aayan636, 2021-03-11:
@premkumar1303 SeeDot can support FastGRNN based algorithms but that is currently under development. If you wish, you can check out the SeeDOT-Face-Detection branch to try it out. A sample code for fastgrnn would be available here: https://github.com/microsoft/EdgeML/blob/SeeDOT-Face-Detection/tools/SeeDot/seedot/compiler/input/fastgrnn.sd 
Take care to update the dimensions and ranges of the input matrices according to your trained model.

As for Raspberry Pi, SeeDot currently does not have a code generator for that, it's only available for Arduino and M4 class devices. You could modify the SeeDot generated Arduino code if you have a Raspberry Pi wrapper. @adityakusupati could you please look into this?

adityakusupati, 2021-03-11:
@premkumar1303, I don't think I used any special wrapper for Raspberry Pi. Any normal C code with proper libraries works on Rpi. IIRC, we used to take the same code and change headers for arm devices and raspberry pi.

I am trying to find the files I used, but again we haven't done anything fancy except the header chnages.

premkumar1303, 2021-03-12:
Ok. If you (@adityakusupati ) could find the files you used, please share. 

Thank you all.

adityakusupati, 2021-03-12:
There should be some pointers or Rpi code in this. https://drive.google.com/file/d/1CfaoyHK-Ny9Bfgui_6r_hlSgZ0JV1BuN/view?usp=sharing

I think I was correct when I said, we didn't change anything except the headers.

premkumar1303, 2021-03-12:
Thank you @adityakusupati. Yes these are the things which I was searching for. They are really helpful.

ShikharJ, 2021-03-13:
@premkumar1303 I'm closing this issue for now. Please feel free to reopen if you have further questions.

closed 2021-03-13 by ShikharJ