microsoft/EdgeML #232: Exception: FastGRNNCUDA is supported only on GPU devices.

mehreenjabeen, 2021-03-21:
Getting error "Exception: FastGRNNCUDA is supported only on GPU devices." upon running kws-demo.py

ShikharJ, 2021-03-21:
@mehreenjabeen I'm assuming your system has a GPU setup? Can you provide the sample command you ran and the full output?

mehreenjabeen, 2021-03-21:
i ran this command: "python kws_demo.py --config_path <path_to_config.json> --model_path <path_to_model.pt> --mean_path <path_to_mean.npy> --std_path <path_to_std.npy>" 
However, I had trained the model with architecture argument set as "-architecture FastGRNNCUDA". Now I am retraining with architecture parameter set to FastGRNN. Hope that resolves the issue

mehreenjabeen, 2021-03-21:
the above thing didnt work(i.e chaging architecture from FastGRNNCUDA to FastGRNN )
So command i ran "python3 kws-demo.py --config_path /home/mehreen/Work/EdgeML/EdgeML/config.json --model_path /home/mehreen/Work/EdgeML/EdgeML/examples/pytorch/FastCells/KWS-training/model/FastGRNN128KeywordSpotter.pt --mean_path $MODEL_DIR/mean.npy --std_path $MODEL_DIR/std.npy"

Output: 
sigmoid tanh
Traceback (most recent call last):
  File "kws-demo.py", line 202, in <module>
    fastgrnn = create_model(config.model, num_filt, 35)
  File "/home/mehreen/Work/EdgeML/EdgeML/examples/pytorch/FastCells/KWS-training/train_classifier.py", line 420, in create_model
    model_config.update_nonlinearity, num_keywords)
  File "/home/mehreen/.local/lib/python3.6/site-packages/edgeml_pytorch/trainer/fastmodel.py", line 59, in __init__
    for l in range(self.num_layers)])
  File "/home/mehreen/.local/lib/python3.6/site-packages/edgeml_pytorch/trainer/fastmodel.py", line 59, in <listcomp>
    for l in range(self.num_layers)])
  File "/home/mehreen/.local/lib/python3.6/site-packages/edgeml_pytorch/graph/rnn.py", line 1121, in __init__
    raise Exception('FastGRNNCUDA is supported only on GPU devices.')
Exception: FastGRNNCUDA is supported only on GPU devices.

ShikharJ, 2021-03-21:
@MJ10 Can you please take a look at this?

ShikharJ, 2021-03-29:
@MJ10 Ping

mehreenjabeen, 2021-03-30:
I am closing this issue as I got to know what the issue was.  In _config.json_ the model architecture was set to FastGRNNCuda because of which it was throwing the error. I changed it to Fastgrnn and it worked!

closed 2021-03-30 by mehreenjabeen