microsoft/EdgeML #237: Question about Bonsai training

sunying2018uchi, 2021-06-01:
Hi,

I am trying to experiment on the bonsai project but find something strange. I used the command `BonsaiTrain -F 256 -C 2 -nT 7291 -nE 2007 -f 0  ./usps2_2 ` where usps2_2 is the directory including the usps dataset (binary classification version which also mentioned in the paper). But I checked the trained params such as Z, I found the dimension is `10*257`.  As we know, the first column of libsvm format data is label column so I expeced the Z should be `10*256` (10 is the projection dimension). Correct me if I am wrong on it. Do you guys meet the same problem? Any help will be appreciated. Thanks!

adityakusupati, 2021-06-01:
Hi,

We add an additional dimension to the matrix to get the bias term in.
Concatenating 1 at the end of feature vector essentially adds bias even
with matmul.

Hope this helps

Aditya

sunying2018uchi, 2021-06-01:
> Hi, We add an additional dimension to the matrix to get the bias term in. Concatenating 1 at the end of feature vector essentially adds bias even with matmul. Hope this helps Aditya
> […](#)

Got it! Thank you so much!

closed 2021-06-01 by sunying2018uchi