microsoft/EdgeML #249: Seedot

mehreenjabeen, 2021-11-25:
Hi, 

I want to generate inference code corresponding to an onnx model. Although  readme file in overview section says "SeeDot takes as input trained floating-point models", I cannot find any argument in the SeeDot-dev.py file which takes trained model as input. 

Please help to resolve this.

ShikharJ, 2021-11-25:
@mehreenjabeen Hi, could you mention which model do you wish to generate the corresponding code for?

mehreenjabeen, 2021-11-25:
Hi shikhar,

I have two onnx models for keyword spotting task based on dnn and gru architecture

ShikharJ, 2021-11-25:
@mehreenjabeen SeeDot/Shiftry might not have the full support for these models, but you can always try. See [this](https://github.com/microsoft/EdgeML/blob/master/tools/SeeDot/README.md) for an example.

mehreenjabeen, 2021-11-25:
I am following  the same link you have shared, but my query is that there is no argument for feeding the trained onnx model to SeeDot-dev.py file.
usage: SeeDot-dev.py [-h] [-a] [-e] [-d] [-m] [-n] [-dt] [-t] [-s] [-sf] [-l] [-lsf] [-tdr] [-o]

ShikharJ, 2021-11-25:
Ah, sorry, please follow the instructions from [here](https://github.com/microsoft/EdgeML/tree/master/tools/SeeDot/seedot/compiler/ONNX). Please note that the generated `seedot_ast.txt` will be the input.sd file that you need. As for dumping the model weights, you can simply do an online search on how to dump weights from a trained ONNX model.

mehreenjabeen, 2021-11-27:
@ShikharJ 
Upon running _compile.sh_ fromhttps://github.com/microsoft/EdgeML/blob/master/tools/SeeDot/seedot/compiler/ONNX/compile.sh, following error appears: `python: can't open file 'create_input.py': [Errno 2] No such file or directory`
 I ran the below command: `./compile.sh ./models/dnn.onnx`

ShikharJ, 2021-11-30:
@mehreenjabeen Sorry for the delayed response, it seems that the SeeDot converter is currently broken, and might need a substantial amount of work. We're planning a new release in February, but till then, I don't see this issue getting resolved.