microsoft/EdgeML #250: How to predict a Test Sample?

affot, 2022-01-18:
Hello,

I am just reviewing the examples for Bonsai and ProtoNN in TF and I'm wondering how to use the model after it has been trained. The Bonsai Class has a getPrediction() method but without any parameters? How can I hand over a new sample to get a prediction?
It would be great to get a feedback from your side.

Best Regards
affot

metastableB, 2022-01-19:
Hi @affot, to obtain the predictions you should first obtain a reference to the prediction operation by 
 ```
    op = protoNNObj.getPredictionsOp() # or 
    op = bonsaiObj.getPredictions()
```

Once you have this, use your text data in the input X placeholder and evaluate this operation to obtain the predictions:

```
sess.run(op, feed_dict ={ X: X_test}) 
```

affot, 2022-01-19:
Hello @metastableB 

thanks a lot for your fast help here!! :-)
For protoNN it works. In the example I can handover for example the first 20 samples of the test set with:
`predictions = sess.run(op, feed_dict ={ X: x_test[0:20]})` and it returns me an array with 20 predicted classes.

With the BonsaiObject I still have a problem. When I do the same here I get an error:

"InvalidArgumentError: You must feed a value for placeholder tensor 'sigmaI' with dtype float [[{{node sigmaI}}]]"
and
"InvalidArgumentError: You must feed a value for placeholder tensor 'sigmaI' with dtype float [[node sigmaI (defined at C:\Users\CK\anaconda3_neu\envs\BT36\lib\site-packages\tensorflow_core\python\framework\ops.py:1748) ]]"

I allready tried `predictions = sess.run(op, feed_dict ={ X: x_test[0:20], sigmaI: 1.0})` but here I get 
"NameError: name 'sigmaI' is not defined" and don't find a solution.
Do you know where is the problem?

metastableB, 2022-01-19:
You will need a reference to the `sigmaI` placeholder in the computation graph. You could try,

```
sigmaI = bonsaiTrainer.sigmaI
predictions = sess.run(op, feed_dict ={ X: x_test[0:20], sigmaI: 1.0})
```

See the declaration of `sigmaI`  [here](https://github.com/microsoft/EdgeML/blob/ef9f8a77f096acbdeb941014791f8eda1c1bc35b/tf/edgeml_tf/trainer/bonsaiTrainer.py#L56).

Also for inference, you should set `sigmaI` to some large value ( `1e10` ) instead of 1.0.

affot, 2022-01-20:
Don Dennis, thanks for your fast help here!!

ShikharJ, 2022-01-24:
Closing this for now, since the query is answered.