microsoft/EdgeML #251: SeeDot-dev.py only working for fastgrnn

affot, 2022-01-24:
Hello,

the SeeDot explanation is working for the Fastgrnn.
But when I transfer the same steps for Bonsai or ProtoNN, the workaround makes some problems.
For example:
`python SeeDot-dev.py -a bonsai -e fixed -d usps10 -n 1 -t arduino -m red_disagree -l info`

It starts with an Error for the assignment in line 227: acc = obj.testingAccuracy with the information, that the Main-Object has no "testingAccuracy".

Is there an easy way to fix this an do the same quantization for bonsai and protoNN?

ShikharJ, 2022-01-24:
@affot The code pertaining to `SeeDot` is slightly unorganised. I've encountered this error before as well, and most of the times, it is because of the datasets and the model files not being in the right directory. Can you just check if there is a `bonsai/usps10` directory in both the `models/` and `datasets/` folders? Ideally, for each algorithm, you should be structuring the `models/bonsai` and `datasets/bonsai` folder in the same manner as the `models/fastgrnn` and `datasets/fastgrnn`.

affot, 2022-01-24:
Hello @ShikharJ 
thanks for your fast reply.
Yes, I checked it several times, I created the same folder/file structures as in the fastgrnn example.

ShikharJ, 2022-01-24:
Hmm, in that case, could you look at `temp/Predictor/output/fixed/` and paste the outputs of `build.txt` and `exec.txt` please?

affot, 2022-01-24:
I don't see no output folder in `temp/Predictor`. There is just an `input` folder?

ShikharJ, 2022-01-24:
Okay, I must apologize for leading you astray. It seems that the nomenclature that is being followed internally is `usps-multiclass` instead of `usps10`. Could you try changing `datasets/bonsai/usps10` to `datasets/bonsai/usps-multiclass` and `models/bonsai/usps10` to `models/bonsai/usps-multiclass` and change the command you're using to `python SeeDot-dev.py -a bonsai -e fixed -d usps-multiclass -n 1 -t arduino -m red_disagree -l info`, please?

affot, 2022-01-25:
I did it, but I get the same Error :-( 

```
========================================
Executing on bonsai fixed usps-multiclass arduino
========================================

INFO:root:-----------------------

INFO:root:Collecting profile data

INFO:root:-----------------------

Traceback (most recent call last):
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/main.py", line 241, in convert
    obj.run()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/compiler/converter/converter.py", line 51, in run
    obj.run(self.source)
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/compiler/converter/quantizer.py", line 252, in run
    self.processModel()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/compiler/converter/quantizer.py", line 190, in processModel
    self.readModel()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/compiler/converter/quantizer.py", line 78, in readModel
    getModelDir(), param.name + ".npy"))
  File "/home/ck/anaconda3/envs/BT36/lib/python3.6/site-packages/numpy/lib/npyio.py", line 422, in load
    fid = open(os_fspath(file), "rb")
FileNotFoundError: [Errno 2] No such file or directory: 'model/bonsai/usps-multiclass/W1.npy'
Traceback (most recent call last):
  File "SeeDot-dev.py", line 270, in <module>
    obj.run()
  File "SeeDot-dev.py", line 164, in run
    self.runMainDriver()
  File "SeeDot-dev.py", line 227, in runMainDriver
    acc = obj.testingAccuracy
AttributeError: 'Main' object has no attribute 'testingAccuracy'
```

ShikharJ, 2022-01-25:
Could you share the `input.sd` inside the `model/bonsai/usps-multiclass`?

affot, 2022-01-25:
```
// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT license.

let W1 = (30, 8) in [-0.1, 0.1] in
let W2 = (8, 64) in [-0.1, 0.1] in
let U1 = (64, 8) in [-0.1, 0.1] in
let U2 = (8, 64) in [-0.1, 0.1] in
let Bg = (1, 64) in [-0.1, 0.1] in
let Bh = (1, 64) in [-0.1, 0.1] in
let FC1 = (64, 20) in [-0.1, 0.1] in
let FC2 = (20, 72) in [-0.1, 0.1] in
let FCBias = (1, 72) in [-0.1, 0.1] in

let X = (210,1) in [-0.1, 0.1] in

let XX = reshape(X, (7, 1, 30), (1, 2)) in

let zeta = 0.15637416 in
let nu = 0.01094334 in

let H = init([1, 64], 0.0) in

let res = loop(i = [0:7], H)
(
	let a = (XX[i] * W1) * W2 in
	let b = (H * U1) * U2 in
	let c = a + b in
	let g = sigmoid(c + Bg) in
	let h = tanh(c + Bh) in
	let H = (g <*> H) + (zeta * (1.0 - g) + nu) <*> h in H
) in

let score = ((res * FC1) * FC2) + FCBias in
argmax(score)
```

ShikharJ, 2022-01-26:
@affot Sorry for the late reply. You're using the DSL code pertaining to `FastGRNN` for running the `Bonsai` code for `usps-multiclass`. The converter is currently broken, and we're trying to fix it in the coming month. You might wanna use the following script instead of your current `input.sd`:

```
let X   = (257, 1)   in [-2.406600, 5.292900] in
let Z   = (14, 257)  in [-0.293206, 0.466991] in
let W   = (3, 10, 14) in [-0.837834, 0.865861] in
let V   = (3, 10, 14) in [-0.950315, 0.847806] in
let T   = (1, 1, 14) in [-0.030703, 0.069962] in
let mean = (14, 1) in [-0.000001, 0.000001] in

let ZX = Z |*| X - mean in

// depth 0
let node0   = 0    in
let W0      = W[node0] * ZX in
let V0      = V[node0] * ZX in
let V0_tanh = tanh(V0) in
let score0  = W0 <*> V0_tanh in

// depth 1
let node1   = (T[node0] * ZX) >= 0? 2 * node0 + 1 : 2 * node0 + 2 in
let W1      = W[node1] * ZX in
let V1      = V[node1] * ZX in
let V1_tanh = tanh(V1) in
let score1  = score0 + W1 <*> V1_tanh in

argmax(score1)
```

Please note that the `model/bonsai/usps-multiclass` directory should contain `X.npy`, `Z.npy` etc as declared above. And you might need to change the provided numerical ranges, based on the dumped weights of the trained `Bonsai` model.

affot, 2022-01-26:
Hello @ShikharJ 
one day delay... So no need to excuse!! :-)
I still have the problem...
What I did:
Copy&Paste the code in the `EdgeML/tools/SeeDot/model/bonsai/usps-multiclass/input.sd`
Copy&Paste all files from `EdgeML/examples/tf/Bonsai/usps10/TFBonsaiResults/20_15_05_24_01_22/SeeDot` (Mean, Sigma, Std, T,V,W,Z) in `EdgeML/tools/SeeDot/model/bonsai/usps-multiclass`

And then:
`python SeeDot-dev.py -a bonsai -e fixed -t arduino -m red_disagree -n 1 -d usps10` -->Error: `python SeeDot-dev.py -a bonsai -e fixed -t arduino -m red_disagree -n 1 -d usps10`
I also tried
`python SeeDot-dev.py -a bonsai -e fixed -t arduino -m red_disagree -n 1 -d usps-multiclass` -->Error: `python SeeDot-dev.py -a bonsai -e fixed -t arduino -m red_disagree -n 1 -d usps10`

ShikharJ, 2022-01-27:
@affot Sorry for the delay once more :) Nearly all of the debugging regarding SeeDot can be done by looking at `temp/Predictor/output` folders. Basically, the second command is correct, but somewhere during the execution of the compiled binary code for the `arduino` platform, some execution error is happening. Could you run the second command and paste the output of the mentioned folder, please?

swapnilsayansaha, 2022-01-27:
I have similar issue. In my case, however, the dataset is a custom dataset which has been properly converted into .npy files. The input array to the fastgrnn model is 550*9.

ShikharJ, 2022-01-27:
@swapnilsayansaha The values in the square brackets are the floating-point ranges of the concerned tensors. So for example, if your `W.npy` has a minimum element of `0.1`, and a maximum element of `0.99`, then the input.sd should look like let `W = (x, y) in [0.1, 0.99] in`

`X` is the input array, and `XX` is just a reshape operation applied to `X` and stored in a separate array. I hope this helps.

swapnilsayansaha, 2022-01-27:
Thanks a lot. That really helped. I have a few more questions.

If X = (550,6), what should be XX be reshaped to?

I can provide some intuition. In the example of Seedot, we have:

```
let X = (256, 1)  in [-2.0, 7.0] in
let XX = reshape(X, (16, 1, 16), (1, 2)) in
```
This comes from the fact the USPS dataset input is a 256*1 array (assuming batch size of 1). The input dim was 16 and hidden dim was 32. The output is binary, which explains (1,2) in the output (?)

In my case: my input dim is [550,6] and hidden dim is 50. the output is (1,9). I was wondering how to properly select the parameters in reshape line of XX.

ShikharJ, 2022-01-27:
@swapnilsayansaha The hidden dimension has got nothing to the shape of X, it is the second dimension of `W` and `U` matrices that determines that. The reshape works in the following manner:

```
let XX = reshape(X, (16, 1, 16), (1, 2)) in
```
The first tuple in the reshape operation depicts the final shape of the reshape output (or simply, the shape of `XX`). The last tuple depicts the order in which the dimension of X should be processed. So in this example, the command semantically signifies "Reshape X into 16 patches of size (1, 16), while traversing X in a row-first fashion". I hope this clears things up.

swapnilsayansaha, 2022-01-27:
Thanks a lot for your help. I had to force the compilation by manually inputting accuracy and it compiled. For some reason obj.testingAccuracy returns empty value.

affot, 2022-01-27:
When I cast `python SeeDot-dev.py -a bonsai -e fixed -t arduino -m red_disagree -n 1 -d usps-multiclass` the Error is 
`========================================
Executing on bonsai fixed usps-multiclass arduino
========================================

Traceback (most recent call last):
  File "SeeDot-dev.py", line 270, in <module>
    obj.run()
  File "SeeDot-dev.py", line 164, in run
    self.runMainDriver()
  File "SeeDot-dev.py", line 225, in runMainDriver
    obj.run()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/main.py", line 839, in run
    return self.runForFixed()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/main.py", line 752, in runForFixed
    res = self.collectProfileData()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/main.py", line 696, in collectProfileData
    execMap = self.predict(config.Encoding.floatt, config.DatasetType.training)
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/main.py", line 260, in predict
    execMap = obj.run()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/predictor.py", line 211, in run
    execMap = self.execute()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/predictor.py", line 185, in execute
    return self.executeForLinux()
  File "/home/ck/Downloads/EdgeML/tools/SeeDot/seedot/predictor.py", line 171, in executeForLinux
    process = subprocess.call(args, stdout=file, stderr=subprocess.STDOUT)
  File "/home/ck/anaconda3/envs/BT36/lib/python3.6/subprocess.py", line 287, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/home/ck/anaconda3/envs/BT36/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/home/ck/anaconda3/envs/BT36/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: './Predictor': './Predictor'
`

and in `temp/Predictor/output/float` I have 2 files (build.txt, exec.txt)

build.txt:
`g++ -c -o debug.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread debug.cpp
g++ -c -o library_fixed.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread library_fixed.cpp
library_fixed.cpp: In function ‘void NormaliseL2(MYINT*, MYINT*, MYINT, MYINT, MYINT, MYINT, MYINT, MYINT)’:
library_fixed.cpp:915:30: warning: suggest parentheses around ‘-’ inside ‘<<’ [-Wparentheses]
  915 |     MYINT yHigh = (1 << shrA - 1);
      |                         ~~~~~^~~
library_fixed.cpp: In function ‘void Sigmoid(MYINT*, MYINT, MYINT, MYINT, MYINT, MYINT, MYINT, MYINT, MYINT*)’:
library_fixed.cpp:961:8: warning: unused variable ‘scale_diff’ [-Wunused-variable]
  961 |  MYINT scale_diff = scale_out / scale_in;
      |        ^~~~~~~~~~
g++ -c -o library_float.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread library_float.cpp
g++ -c -o main.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread main.cpp
main.cpp: In function ‘int main(int, char**)’:
main.cpp:500:33: warning: comparison with string literal results in unspecified behavior [-Waddress]
  500 |  if ((argc == 6) && (argv[5] == "False"))
      |                                 ^~~~~~~
main.cpp:142:8: warning: unused variable ‘epsilon’ [-Wunused-variable]
  142 |  float epsilon = 0.00001;
      |        ^~~~~~~
g++ -c -o profile.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread profile.cpp
g++ -c -o seedot_fixed.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread seedot_fixed.cpp
g++ -c -o seedot_float.o -Wall -Wno-narrowing -p -g -fPIC -O3 -std=c++11 -pthread seedot_float.cpp
In file included from seedot_float.cpp:9:
model_float.h:84:1: error: too many initializers for ‘const float [3][10][14]’
   84 | };
      | ^
model_float.h:157:1: error: too many initializers for ‘const float [3][10][14]’
  157 | };
      | ^
model_float.h:163:1: error: too many initializers for ‘const float [1][1][14]’
  163 | };
      | ^
In file included from seedot_float.cpp:9:
model_float.h:166:3328: error: too many initializers for ‘const float [14][1]’
  166 |  -0.991950f,  -0.972569f,  -0.930964f,  -0.853716f,  -0.734950f,  -0.580235f,  -0.392483f,  -0.226989f,  -0.217470f,  -0.351633f,  -0.517160f,  -0.670813f,  -0.799098f,  -0.893281f,  -0.953961f,  -0.985741f,  -0.982065f,  -0.939628f,  -0.848189f,  -0.687138f,  -0.468550f,  -0.236652f,  -0.010466f,  0.169037f,  0.166962f,  0.000575f,  -0.195308f,  -0.405139f,  -0.621117f,  -0.796838f,  -0.910481f,  -0.970707f,  -0.972356f,  -0.908924f,  -0.774088f,  -0.553005f,  -0.301348f,  -0.108794f,  0.024146f,  0.121127f,  0.093866f,  -0.026314f,  -0.130687f,  -0.278695f,  -0.508489f,  -0.734396f,  -0.884430f,  -0.962162f,  -0.964207f,  -0.884020f,  -0.720834f,  -0.480033f,  -0.259240f,  -0.159720f,  -0.134882f,  -0.110839f,  -0.142934f,  -0.190815f,  -0.187280f,  -0.251017f,  -0.456327f,  -0.700435f,  -0.873188f,  -0.961192f,  -0.956769f,  -0.861720f,  -0.683815f,  -0.451380f,  -0.276722f,  -0.242718f,  -0.272671f,  -0.275193f,  -0.286166f,  -0.277784f,  -0.224064f,  -0.255743f,  -0.443262f,  -0.684031f,  -0.864659f,  -0.960294f,  -0.947782f,  -0.839320f,  -0.659494f,  -0.448801f,  -0.304465f,  -0.285991f,  -0.316657f,  -0.306787f,  -0.287803f,  -0.262377f,  -0.227436f,  -0.280945f,  -0.461721f,  -0.681623f,  -0.853369f,  -0.953936f,  -0.935957f,  -0.817802f,  -0.645808f,  -0.460782f,  -0.334343f,  -0.304040f,  -0.302037f,  -0.256574f,  -0.214390f,  -0.203471f,  -0.218826f,  -0.311599f,  -0.486943f,  -0.678994f,  -0.835285 [... cut ...]                                                                                                                                                                                                                                                                                                                                                                                                                                              ^
make: *** [Makefile:49: seedot_float.o] Fehler 1`

exec.txt is empty.

swapnilsayansaha, 2022-01-27:
Can confirm I have the same error.

ShikharJ, 2022-01-27:
I see, `model_float.h` is used for storing the `.npy` weight files into C++ format. From the error, it seems that the dimensionality of the `.npy` files, and the description in the `input.sd` for the same does not match. Can you please check?

affot, 2022-01-28:
This is correct, the dimensions of the `.npy `Files and the entries in `input.sd` don't match.
When I load the files and print the .shape with `print("Z" , np.fromfile('Z.npy').shape)` I get:
```
Z (1301,)
W (366,)
V (366,)
T (31,)
```

`X.npy` isn't even existing.

If you want to understand what happened here, I just followed the example to train the bonsai algorithm, nothing has been changed on my side.

ShikharJ, 2022-01-28:
@affot Okay, this seems like a major issue then. I'll need time to look into it. Sorry for the inconvenience.

affot, 2022-01-28:
No Problem, thanks a lot for taking care!

swapnilsayansaha, 2022-01-28:
What I understood from trying to run anything apart from the USPS-10 tutorial given in the repo, even training the model for 1 epoch less (the USPS-10 fastgrnn model) causes Seedot compilation to fail (sometimes Seedot says X cannot be quantized properly, sometimes it says test accuracy object not found, sometimes it can't find quantization functions). I think a lot of things are internally hard-coded to work for the specific example given. Maybe a more general tutorial on what things to modify for a custom model and a custom dataset might help.