microsoft/EdgeML #252: The model of DROCC seems missing

NoThinG-qh, 2022-01-25:
Thank you for the contribution to one class classification. DROCC is an excellent algorithm, but it seems that only the trainer of DROCC was found in this code repository , I could not find the model of DROCC. Can the model be published?

ShikharJ, 2022-01-26:
@NoThinG-qh Thanks for your interest in DROCC. I think trained models have license conflicts with the repository, so we don't publish them. One can always use the trainer code for generating the models again.

SachinG007, 2022-01-27:
@NoThinG-qh you can use the CLI commands here to generate trained models https://github.com/microsoft/EdgeML/tree/master/examples/pytorch/DROCC

ShikharJ, 2022-01-27:
Closing for now.

closed 2022-01-27 by ShikharJ