microsoft/EdgeML #253: How to get predictions from trained model for FastCells in Tensorflow

swapnilsayansaha, 2022-01-28:
For Bonsai and ProtoNN, we can convert the trained TF model to tflite and perform inference on new data. I was wondering if something similar exists for FastCells in TF, where we could load the pre-trained model parameters and perform inference (maybe not through TFLite but through TF directly)

ShikharJ, 2022-01-29:
@adityakusupati Can you please take a look?

adityakusupati, 2022-01-29:
Hi,

The same sort of inference should work for FastCells too however, this is
all in tf 1.0 which has limited support.

You can use see dot to automatically compile low level code files for
inference.

Hope this helps.

swapnilsayansaha, 2022-01-30:
Seedot has some issues converting models trained on custom dataset #251 

Anyways I exposed the predictions from the training script:

In fastTrainer.py

```
 testAcc, testLoss, s_predictions, s_truth = sess.run([self.accuracy, self.lossOp, self.Y_output, self.Y_truth], feed_dict={
                                         self.X: Xtest, self.Y: Ytest})

            print('s_predictions:', s_predictions) #model predictions
            print('s_truth:', s_truth) #ground truth labels
```

in the init() function:
```
        self.Y_output = tf.argmax(self.predictions, 1)
        self.Y_truth = tf.argmax(self.Y, 1)
```

closed 2022-01-30 by swapnilsayansaha